espressif/esptool #969: Unsupported Command Errors when attempting to use esptools to flash esp8266 chip using a Windows PC

CdreKidd, 2024-04-07:
### Operating System

Windows 10

### Esptool Version

4.7.0

### Python Version

3.9.5

### Chip Description

ESP 8266

### Device Description

Makerfabs Wifi mated to Arduino mega 

### Hardware Configuration

No

### How is Esptool Run

Windows Command Line

### Full Esptool Command Line that Was Run

esptool.py --chip esp8266 --baud 38400 chip_id

### Esptool Output

```shell
Traceback (most recent call last):
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\Scripts\esptool.py-script.py", line 33, in <module>
    sys.exit(load_entry_point('esptool==4.7.0', 'console_scripts', 'esptool.py')())
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool\__init__.py", line 1139, in _main
    main()
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool\__init__.py", line 712, in main
    esp = esp or get_default_connected_device(
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool\__init__.py", line 1024, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool\loader.py", line 678, in connect
    last_error = self._connect_attempt(reset_strategy, mode)
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool\loader.py", line 580, in _connect_attempt
    self.sync()
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool\loader.py", line 496, in sync
    val, _ = self.command(
  File "C:\Users\pgfow\AppData\Local\Programs\Python\Python39\lib\site-packages\esptool\loader.py", line 450, in command
    raise UnsupportedCommandError(self, op)
esptool.util.UnsupportedCommandError: Invalid (unsupported) command 0x8
```

### More Information

This, or very similar errors occur every time esptools.py is run with various command line commands.  I get the same errors when using esptools ver 3.3.3 and 4.5.1 in .exe and .py format

### Other Steps to Reproduce

Other DCC EX users report they are unable to run esptools successfully on windows pcs to flash esp8266 chips with the same errors recorded. I have chosen this particular example as it has the least sources of potential error.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

jakub-kocka, 2024-04-08:
Hello @CdreKidd,

Sorry to hear that you have trouble with flashing. Unfortunately, we are not able to reproduce your issue on a Windows machine with an ESP8266 chip.

Can you tell me how did you install the esptool? Did you follow the [guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html) or as the part of ESP-IDF framework?

Also, I have tested it with the v4.7.0 exe release (https://github.com/espressif/esptool/releases/tag/v4.7.0)

    esptool.exe --chip esp8266 --baud 38400 chip_id
    
And it works as expected.

Maybe this could be an issue with your PC's drivers, can you try to update them?

jakub-kocka, 2024-04-09:
@CdreKidd,
USB drivers.

I have noticed that the Python module invoked for esptool seems to be not just esptool.py but `esptool.py-script.py` (from the second line of the output you have provided). I would suggest re-install esptool (using pip or remove the directory and install it again).

If you need to verify that the esptool runs on your machine you can follow up with the installation guide to ["Flashing firmware"](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html#id4) or just use the `chip_id` command as you have mentioned.

    python -m esptool --chip esp8266 --baud 38400 chip_id

But if I understand correctly, you are trying to flash the ESP8266 not directly with the esptool but with the DCC EX? 

Maybe this will be helpful (https://dcc-ex.com/support/wifi-at-version.html#gsc.tab=0).

CdreKidd, 2024-04-09:
Thanks - the 4.7.0 was installed using pip - I will try it again - would it be better installed under python 12 or 3.9?  

Thanks for the link to DCC EX pages - I have been following their advice, hence the 4.5.1 setup I used at first - however, I am seeing the same issue other DCC EX users report and I note this comment on their website - see below. Trying to get to the bottom of why it appears to be a Windows specific issue (as a Window user)

 [Correcting the Makerfabs ESP8266 WiFi Shield With an Arduino Mega](https://dcc-ex.com/support/wifi-at-version.html#id34)[](https://dcc-ex.com/support/wifi-at-version.html#correcting-the-makerfabs-esp8266-wifi-shield-with-an-arduino-mega)

Warning

**The instructions on the following page have been proven to work on Linux PCs, so will probably work on macOS PCs (Apple), however attempts to get them to work on Microsoft Windows PCs have so far been unsuccessful.**

Flashing the ESP8266 WiFi Shield [using an Arduino Mega is covered here](https://dcc-ex.com/support/makerfabs-update-at-version-with-mega.html).

jakub-kocka, 2024-04-09:
> Thanks - the 4.7.0 was installed using pip - I will try it again - would it be better installed under python 12 or 3.9?
>
> Thanks for the link to DCC EX pages - I have been following their advice, [...]

Both versions of Python should be fine, but I would suggest using a newer one. Also, on Windows, it is better to download it from the official [Python website](https://www.python.org/downloads/) rather than using the MS Store version (just a tip if you were in doubt).

jakub-kocka, 2024-04-11:
We are closing this because it is not an issue with the esptool.

jakub-kocka, 2024-04-15:
It appears from the above discussion that the problem does not stem from the esptool. The documentation for DCC EX indicates that the Makerfabs ESP8266 board for DCC EX has encountered some issues, along with suggested steps to resolve them.

Therefore, we are marking this issue as resolved. @CdreKidd, feel free to reopen it if necessary.

closed 2024-04-15 by radimkarnis