espressif/esptool #971: esptool: error: unrecognized arguments: --force --encrypt

rsVenotec, 2024-04-10:
### Operating System

Windows 11

### Esptool Version

v4.7.0  bff9334

### Python Version

Python 3.11.9

### Full Esptool Command Line that Was Run

esptool.exe --port "COM6" --baud 921600  --no-stub --force erase_flash

### Esptool Output

```shell
esptool: error: unrecognized arguments: --force
```

### What is the Expected Behaviour?

Hi all, i want to re Progrogramm an esp32-s3 with secure boot explained on: [circuitdigest.com](https://circuitdigest.com/article/how-to-secure-esp32-firmware-and-flash-memory-using-esp-idf-framework)
or delete it completely to reprogramme it.

In the documents their for is the --force option described

https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html
but like descripted esptool does not found any --Force option for esp32s3
With `esptool.exe --port "COM6" --baud 921600  --no-stub erase_flash`
Esptool report: 
**A fatal error occurred: ESP32-S3 ROM does not support function erase_flash.**

So for me their are some options missing. Or do i have overlooked something?

dobairoland, 2024-04-10:
Please see `esptool.py --help`. There is no `--force` there. It is not  a general esptool argument

Also see `esptool.py write_flash --help`. `--force` is mentioned there. Therefore, force can be applied for writing flash and the argument must be after `write_flash`.

The second issue is that you use `--no-stub`. As it is explained in the error message, `erase_flash` cannot be invoked with `--no-stub`.

closed 2024-04-10 by dobairoland

radimkarnis, 2024-04-10:
The `--force` argument is also available for `erase_flash`, but as @dobairoland explains, you need to put it after the command: `esptool.exe --port "COM6" --baud 921600 erase_flash --force`

rsVenotec, 2024-04-10:
Thank you very much, I wouldn't have thought of that. That Trace belongs to write flash!

Unfortunately I still couldn't programme the ESP: 
I have tried the following:

**_esptool.exe --no-stub --port "COM6" --baud 921600 erase_flash --force_**
gives: 
```
esptool.py v4.7.0
Serial port COM6
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 921600
Changed.
Enabling default SPI flash mode...
Erasing flash (this may take a while)...

A fatal error occurred: ESP32-S3 ROM does not support function erase_flash
```

Trying to upload an empty sketch :
**esptool.exe" --no-stub --chip esp32s3 --port "COM6" --baud 921600  --before default_reset --after hard_reset write_flash --force --encrypt -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x10000 "sketch_apr10a.ino.bin"**
gives: 
```
esptool.py v4.7.0
Serial port COM6
Connecting...
Chip is ESP32-S3 in Secure Download Mode
Changing baud rate to 921600
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00010000 to 0x00048fff...

WARNING: - compress and encrypt options are mutually exclusive
Will flash sketch_apr10a.ino.bin uncompressed
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01640000: Unknown result)
```

Do you have some new ideas to fix?

dobairoland, 2024-04-10:
Please read the documentation for those arguments and don't use "force" without understanding everything. That is a good way to permanently damage your device. 

In both cases the error message is clear and points to which arguments were use wrongly. Please examine and understand all of them.

dobairoland, 2024-04-15:
For the sake of completeness, here are our official security guides: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/index.html (they include the appropriate commands to run)

I want to point out once more that running esptool commands enabling security features without understanding all arguments will most probably lead to damaged devices. So please follow those guides.