espressif/esptool #973: Can't Flash ESP32-S2 on Mac

Doyle4, 2024-04-17:
### Operating System

macOS Catalina  - MBP 2012 Non Retina

### Esptool Version

3.2

### Python Version

3.10.2

### Chip Description

ESP32-S2

### Device Description

USB-C

### Hardware Configuration

Nope

### How is Esptool Run

Terminal

### Full Esptool Command Line that Was Run

esptool.py --port /dev/cu.usbmodem01 --baud 921600 flash_id

### Esptool Output

```shell
Last login: Wed Apr 17 10:47:43 on ttys000
doyle@Macbook-Pro ~ % esptool.py --port /dev/cu.usbmodem01 --baud 921600 flash_id
esptool.py v3.2
Serial port /dev/cu.usbmodem01
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
doyle@Macbook-Pro ~ %
```

### More Information

Im pretty much a noob at this sorry, I can see there has been an update which I downloaded, but I have no idea what to do with the files.

### Other Steps to Reproduce

ESPtools works fine on Windows 10 for me using the program to flash, I use Bootcamp on Mac and sometimes I'm unable to stop what I'm doing to boot into Windows

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-04-17:
Hi @Doyle4,

please follow the [installation instructions](https://docs.espressif.com/projects/esptool/en/latest/esp32s2/installation.html) to get the latest version of esptool.py and try again. Let me know if your issues persist.

radimkarnis, 2024-04-22:
I am closing this issue. Please try to update, follow the troubleshooting guide, and open a new issue if necessary.

closed 2024-04-22 by radimkarnis

Doyle4, 2024-04-22:
Apologies I didn't get back, had a fair bit on.

Still not working sadly, will use Windows ESPtool for now.