espressif/esptool #979: A fatal error occurred: Invalid head of packet (0x6F)

mbastida123, 2024-05-22:
### Operating System

Windows 11

### Esptool Version

esptool.py v3.3.4-dev

### Python Version

Python 3.11.2

### Full Esptool Command Line that Was Run

esptool.py --baud 38400 read_flash 0xB90000 4194304 test.bin

### Esptool Output

```shell
esptool.py v3.3.4-dev
Found 4 serial ports
Serial port COM6
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 74:4d:bd:90:a6:18
Uploading stub...
Running stub...
Stub running...
458752 (10 %)
A fatal error occurred: Invalid head of packet (0x6F): Possible serial noise or corruption.
```

### What is the Expected Behaviour?

Before you tell me this is an issue with the device or the connection:

I have tried on two different ESP32 boards and both of them return the error at the same progress 458752 (10 %).
Moreover, both of the boards flash just fine, the issue is with the read_flash command.

radimkarnis, 2024-05-22:
Hello @mbastida123,

1) Can you please try the latest version of esptool (`v4.8.dev3`), or is this not an option for you?
2) Are you connecting to the ESP32-S3 chip with a USB-to-UART bridge or with the internal USB-Serial/JTAG peripheral? If your board has two ports, does the issue persist when using the other port?
3) Can you try running the command with the `--no-stub` option? (`esptool.py --no-stub ...`

Please let me know if any of these made a difference!

mbastida123, 2024-05-22:
> Hello @mbastida123,
>
> 1. Can you please try the latest version of esptool (`v4.8.dev3`), or is this not an option for you?
> 2. Are you connecting to the ESP32-S3 chip with a USB-to-UART bridge or with [...]

1. Looking at the github repo the latest version is 4.7: https://github.com/espressif/esptool/releases
2. I'm using ESP-PROG. The port I'm using (COM6) is the only one that works
3. Another error:

> esptool.py --no-stub read_flash 0xB90000 4194304 test.bin
> esptool.py v3.3.4-dev
> Found 4 serial ports
> Serial port COM6
> Connecting....
> Detecting chip type... ESP32-S3
> Chip is ESP32-S3 (revision v0.2)
> Features: WiFi, BLE
> Crystal is 40MHz
> MAC: 74:4d:bd:90:a5:fc
> Enabling default SPI flash mode...
> 
> A fatal error occurred: Failed to read flash block (result was 01090000: CRC or checksum was invalid)

radimkarnis, 2024-05-22:
You can get a `v4.8.dev3` pre-release version on PyPI: https://pypi.org/project/esptool/#history. But `v4.7` should be good as well. It's just to see if a more recent version of esptool makes a difference

mbastida123, 2024-05-22:
> You can get a `v4.8.dev3` pre-release version on PyPI: https://pypi.org/project/esptool/#history. But `v4.7` should be good as well. It's just to see if a more recent version of esptool makes a difference

Tested in 4.7 and it works!

Thank you :)

One question though: All the esp tools where downloaded and installed automatically via the vscode extension. What would then be the correct way of updating said tools? Can I just update them manually? Is there a magic button in the vscode extension that does that?

radimkarnis, 2024-05-22:
Are you speaking about the ESP-IDF VSCode extensions? If so, it uses an installation of ESP-IDF (which comes bundled with the necessary tools - including esptool.py). So one option is to update IDF.

Or if you are talking about PlatformIO extension, then you need to update the esp32 core.

mbastida123, 2024-05-22:
> Are you speaking about the ESP-IDF VSCode extensions? If so, it uses an installation of ESP-IDF (which comes bundled with the necessary tools - including esptool.py). So one option is to update IDF.
> 
> Or if you are talking about PlatformIO extension, then you need to update the esp32 core.
I'm using ESP-IDF VSCode extension
So this means that if for example I'm at espressif v4.4.7 I would have to update to V5.1 for example? I don't really see how this would also update the tools...

mbastida123, 2024-05-22:
> > So this means that if for example I'm at espressif v4.4.7 I would have to update to V5.1 for example? I don't really see how this would also update the tools...
> 
> The tools are part of the installation. ESP-IDF v4.4.* uses esptool.py v3.* under the hood, while IDF v5.* uses esptool v4.*. Updating ESP-IDF means the necessary tools get updated as well.

Understood, thank you! :)

closed 2024-05-22 by mbastida123