espressif/esptool #983: Invalid head of packet (0x45): Possible serial noise or corruption...ESP32-C3-MINI

MahadiHasantauhid, 2024-06-15:
### Operating System

Windows11

### Esptool Version

esptool.py v4.7.0

### Python Version

Python 3.12.3

### Chip Description

ESP32-C3-MINI-1

### Device Description

The ESP32 C3 MINI module is connected to FTDI Serial to USB Adapter. I was having BROWNOUT RESET while using the 3V pin from the adapter to Power the module. Now I'm using the 5V pin from the adapter to power up mini module, it seems that the BROWNOUT RESET is not happening during reset. But when I try to flash any program I used a switch to pull down the GPIO9 pin to LOW to the enter the bootmode`rst:0x1 (POWERON),boot:0x5 (DOWNLOAD(USB/UART0/1))
waiting for download
`
The message I see during Normal Boot mode using Tera Term as follows  `ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x18c8
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2d6c
entry 0x403ce000
I (31) boot: ESP-IDF qa-test-v4.3.3-20220423 2nd stage bootloader
I (31) boot: compile time 07:45:53
I (32) boot: chip revision: 4
I (35) boot_comm: chip revision: 4, min. bootloader chip revision: 3
I (42) boot.esp32c3: SPI Speed      : 40MHz
I (46) boot.esp32c3: SPI Mode       : DIO
I (51) boot.esp32c3: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 nvs              WiFi data        01 02 00010000 0000e000
I (95) boot:  3 at_customize     unknown          40 00 0001e000 00042000
I (102) boot:  4 ota_0            OTA app          00 10 00060000 001d0000
I (110) boot:  5 ota_1            OTA app          00 11 00230000 001d0000
I (117) boot: End of partition table
I (121) boot_comm: chip revision: 4, min. application chip revision: 3
I (129) esp_image: segment 0: paddr=00060020 vaddr=3c140020 size=2a620h (173600) map
I (175) esp_image: segment 1: paddr=0008a648 vaddr=3fc91200 size=03c6ch ( 15468) load
I (179) esp_image: segment 2: paddr=0008e2bc vaddr=40380000 size=01d5ch (  7516) load
I (183) esp_image: segment 3: paddr=00090020 vaddr=42000020 size=133ab0h (1260208) map
I (465) esp_image: segment 4: paddr=001c3ad8 vaddr=40381d5c size=0f3c4h ( 62404) load
I (480) esp_image: segment 5: paddr=001d2ea4 vaddr=50000000 size=00014h (    20) load
I (481) esp_image: segment 6: paddr=001d2ec0 vaddr=50000018 size=00010h (    16) load
I (492) boot: Loaded app from partition at offset 0x60000
I (492) boot: Disabling RNG early entropy source...
module_name:MINI-1
max tx power=78,ret=0
`
I'm trying to flash`MINI-1 factory_param.bin` to the module but unfortunately I see the same "**Invalid head of packet (0x45): Possible serial noise or corruption**" I used VSCode IDE, ESP Flasher TOOL. But same issue persist. 

### Hardware Configuration

The Module is connected to STM32 Via Uart.

### How is Esptool Run

VS Code

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-06-17:
Hi @MahadiHasantauhid,
please provide a full log from esptool.py (the one ending with `Invalid head of packet (0x45): Possible serial noise or corruption`). The normal boot log is not relevant to this issue. Thank you.

MahadiHasantauhid, 2024-06-17:
I can provide several but the most recurring one is `Invalid head of packet (0x45): Possible serial noise or corruption`. As follows:
`Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 15:56:17,242][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.
no log file output ...
.
Uploading stub...
Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 15:57:01,242][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.
no log file output ...
test offset :  0 0x0
case ok
test offset :  0 0x0
case ok
.
Uploading stub...
Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 16:01:20,312][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.
no log file output ...
test offset :  0 0x0
case ok
...........................................................................................................
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 16:02:20,221][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.
no log file output ...
test offset :  0 0x0
case ok
.....................................
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 16:04:41,219][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.
no log file output ...
test offset :  0 0x0
case ok
[2024-06-04 16:19:54,898][ESP8266Loader_spi[1]][espDownloader.py][line:773][ERROR]:
no log file output ...
test offset :  0 0x0
case ok
........................................................................
Uploading stub...
Running stub...
Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 16:20:15,669][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.
no log file output ...
test offset :  0 0x0
case ok
........................................................................
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Invalid head of packet (0x45): Possible serial noise or corruption.
[2024-06-04 16:24:55,660][ESP8266Loader_spi[1]][espDownloader.py][line:782][ERROR]: ESP32C3 Chip stub error esp_stub_and_set_baud.
no log file output ...
test offset :  0 0x0
case ok
.
[2024-06-04 17:00:20,192][ESP8266Loader_spi[1]][espDownloader.py][line:592][ERROR]: Chip sync error
[2024-06-04 17:00:20,198][ESP8266Loader_spi[1]][espDownloader.py][line:777][ERROR]: ESP32C3 Chip sync error esp_sync_blocking.
no log file output ...
test offset :  0 0x0
case ok
.
[2024-06-04 17:04:47,657][ESP8266Loader_spi[1]][espDownloader.py][line:592][ERROR]: Chip sync error
[2024-06-04 17:04:47,657][ESP8266Loader_spi[1]][espDownloader.py][line:777][ERROR]: ESP32C3 Chip sync error esp_sync_blocking.
no log file output ...`
this log is taken from **flash_download_tool_3.9.6**

radimkarnis, 2024-06-17:
Could you please try to use esptool.py directly?

If you have Python installed on your system, it should be as easy as:
```
pip install esptool
esptool write_flash factory_param.bin
```

For full installation instructions, please look [here](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/installation.html).

There might be a bug or a regression in the `flash_download_tool` app, or it is using an old version of esptool.py. The only way we can be sure you are using the latest release with all of the recent bug fixes is to use the Python package (or at least the latest Windows executable: [esptool-v4.7.0-win64.zip](https://github.com/espressif/esptool/releases/download/v4.7.0/esptool-v4.7.0-win64.zip))

MahadiHasantauhid, 2024-06-17:
`esptool.py v4.7.0
Serial port COM5
Connecting.........
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: ec:da:3b:a5:bc:c4
Uploading stub...

A fatal error occurred: Invalid head of packet (0x45): Possible serial noise or corruption.` This is the error I kept receiving for the command `esptool --chip esp32c3 --port COM5 --baud 115200 write_flash 0x0 factory_MINI-1.bin` I checked also for baud rate 9600.

radimkarnis, 2024-06-17:
Thank you. The tool seems to fail when the stub flasher is being uploaded. You can run the same command with the `--no-stub` option after `esptool.py` to skip this part and see if this helps.

Also, adding the `--trace` option will show us more diagnostic information!

MahadiHasantauhid, 2024-06-18:
`esptool.py v4.7.0
Serial port COM5
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.030 Read 1 bytes: 42
TRACE +0.000 Read invalid data: 42
TRACE +0.000 Remaining data in serial buffer:
    75696c643a466562 2020372032303231 | uild:Feb  7 2021
    0d0a7273743a3078 312028504f574552 | ..rst:0x1 (POWER
    4f4e292c626f6f74 3a30786420285350 | ON),boot:0xd (SP
    495f464153545f46 4c4153485f424f4f | I_FAST_FLASH_BOO
    54290d0a53504957 503a307865650d0a | T)..SPIWP:0xee..
    6d6f64653a44494f 2c20636c6f636b20 | mode:DIO, clock
    6469763a320d0a6c 6f61643a30783366 | div:2..load:0x3f
    6364363130302c6c 656e3a3078313863 | cd6100,len:0x18c
    380d0a6c6f61643a 3078343033636530 | 8..load:0x403ce0
    30302c6c656e3a30 783864340d0a6c6f | 00,len:0x8d4..lo
    61643a3078343033 64303030302c6c65 | ad:0x403d0000,le
    6e3a307832643663 0d0a656e74727920 | n:0x2d6c..entry
    3078343033636530 30300d0a1b5b303b | 0x403ce000...[0;
    33326d4920283331 2920626f6f743a20 | 32mI (31) boot:
    4553502d49444620 71612d746573742d | ESP-IDF qa-test-
    76342e332e332d32 3032323034323320 | v4.3.3-20220423
    326e642073746167 6520626f6f746c6f | 2nd stage bootlo
    616465721b5b306d 0d0a1b5b303b3332 | ader.[0m...[0;32
    6d49202833312920 626f6f743a20636f | mI (31) boot: co
    6d70696c65207469 6d652030373a3435 | mpile time 07:45
    3a35331b5b306d0d 0a1b5b303b33326d | :53.[0m...[0;32m
    4920283332292062 6f6f743a20636869 | I (32) boot: chi
    7020726576697369 6f6e3a20341b5b30 | p revision: 4.[0
    6d0d0a1b5b303b33 326d492028333529 | m...[0;32mI (35)
    20626f6f745f636f 6d6d3a2063686970 |  boot_comm: chip
    207265766973696f 6e3a20342c206d69 |  revision: 4, mi
    6e2e20626f6f746c 6f61646572206368 | n. bootloader ch
    6970207265766973 696f6e3a20331b5b | ip revision: 3.[
    306d0d0a1b5b303b 33326d4920283432 | 0m...[0;32mI (42
    2920626f6f742e65 7370333263333a20 | ) boot.esp32c3:
    5350492053706565 642020202020203a | SPI Speed      :
    2034304d487a1b5b 306d0d0a1b5b303b |  40MHz.[0m...[0;
    33326d4920283436 2920626f6f742e65 | 32mI (46) boot.e
    7370333263333a20 535049204d6f6465 | sp32c3: SPI Mode
    202020202020203a 2044494f1b5b306d |        : DIO.[0m
    0d0a1b5b303b3332 6d49202835312920 | ...[0;32mI (51)
    626f6f742e657370 333263333a205350 | boot.esp32c3: SP
    4920466c61736820 53697a65203a2034 | I Flash Size : 4
    4d421b5b306d0d0a 1b5b303b33326d49 | MB.[0m...[0;32mI
    202835362920626f 6f743a20456e6162 |  (56) boot: Enab
    6c696e6720524e47 206561726c792065 | ling RNG early e
    6e74726f70792073 6f757263652e2e2e | ntropy source...
    1b5b306d0d0a1b5b 303b33326d492028 | .[0m...[0;32mI (
    36312920626f6f74 3a20506172746974 | 61) boot: Partit
    696f6e205461626c 653a1b5b306d0d0a | ion Table:.[0m..
    1b5b303b33326d49 202836352920626f | .[0;32mI (65) bo
    6f743a202323204c 6162656c20202020 | ot: ## Label
    2020202020202020 5573616765202020 |         Usage
    2020202020202054 797065205354204f |        Type ST O
    6666736574202020 4c656e6774681b5b | ffset   Length.[
    306d0d0a1b5b303b 33326d4920283732 | 0m...[0;32mI (72
    2920626f6f743a20 2030206f74616461 | ) boot:  0 otada
    7461202020202020 202020204f544120 | ta          OTA
    6461746120202020 2020202020303120 | data         01
    3030203030303064 3030302030303030 | 00 0000d000 0000
    323030301b5b306d 0d0a1b5b303b3332 | 2000.[0m...[0;32
    6d49202838302920 626f6f743a202031 | mI (80) boot:  1
[... 218 lines cut ...]
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04006f50311b00000000c0
TRACE +0.000 Received full packet: 010a04006f50311b00000000
TRACE +0.019 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.002 Write 10 bytes: c00014000000000000c0
TRACE +0.002 Read 1 bytes: c0
TRACE +0.000 Read 22 bytes:
    011418006f50311b 0000000000000000 | ....oP1.........
    0000000c0500                      | ......
TRACE +0.012 Read 1 bytes: 00
TRACE +0.000 Read 10 bytes: 000300000000000000c0
TRACE +0.000 Received full packet:
    011418006f50311b 0000000000000000 | ....oP1.........
    0000000c05000000 0300000000000000 | ................
TRACE +0.017 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=7cf0cd3f
TRACE +0.000 Write 14 bytes: c0000a0400000000007cf0cd3fc0
TRACE +0.122 Read 1 bytes: 45
TRACE +0.000 Read invalid data: 45
TRACE +0.000 Remaining data in serial buffer:
    53502d524f4d3a65 7370333263332d61 | SP-ROM:esp32c3-a
    7069312d32303231 303230370d0a     | pi1-20210207..

A fatal error occurred: Invalid head of packet (0x45): Possible serial noise or corruption.` For the Command `esptool --chip esp32c3 --port COM5 --baud 115200 --no-stub --trace write_flash 0x0 factory_firmware.bin
`

radimkarnis, 2024-06-18:
Your chip is resetting in the middle of the flashing process:

```
 53502d524f4d3a65 7370333263332d61 | SP-ROM:esp32c3-a
 7069312d32303231 303230370d0a     | pi1-20210207..
```
    
The `ESP-ROM:esp32c3-api1-20210207` snippet is a part of the [boot log](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html#boot-log) - that appears after a reset.

Now, we don't see the reason why the reset is happening. It could be an insufficient power supply, a faulty unit, or a stray watchdog resetting the chip. 

To find out why this happens, we need to capture the reset reason, which is mentioned later in the boot log (e.g. `rst:0x1 (POWERON_RESET)`).  One way to do this is to try flashing over the internal USB-Serial/JTAG peripheral and listen over your FTDI Serial to USB Adapter at the same time - here you should see the full output (if the issue still happens when USB-Serial/JTAG is used).

MahadiHasantauhid, 2024-06-18:
> Your chip is resetting in the middle of the flashing process:
>
> ```
>  53502d524f4d3a65 7370333263332d61 | SP-ROM:esp32c3-a
>  7069312d32303231 303230370d0a     | pi1-20210207..
> ```
>
> The [...]
@radimkarnis Thank you very much for the responses so far. 
I was experiencing BROWNOUT_RESET while the module was powered 3V from the FTDI Adapter, But now I'm giving 5V just to get rid of that issue. For a time being BROWNOUT_RESET is averted, I also checked the insufficient power issue from the Manual where it is mentioned- VDD3P3_CPU,VDD3P3_RTC, VDDA1 and VDDA2 should have more than **2.7V Threshold Voltage** in order to avoid BROWNOUT_RESET. I will change the USB Cable as well to see whether it is not an issue(Length=1meter). 

Unfortunately, this module is attached to another host MCU by Uart connection. I'm using TXDO, RXDO (Pin 30 &31) to flash the firmware into it. FTDI CTS connected to GPIO9 and EN is left floating to get the DOWNLOAD BOOT MODE. But I never had this **rst:0x1 (POWERON_RESET)** mode till now while checking with TERA TERA or PUTTY, it is Always **rst:0x1 (POWERON)** while on download mode, is it similar type definition? 

it keeps continuously coming to serial terminal both in Teraterm, PUTTY and VS Code. Is it normal to get the messages continuously on terminal or just once while it resets? But I suspected that the Error type is related to **chip reset while flashing** according to the Troubleshooting guide.

MahadiHasantauhid, 2024-06-20:
@radimkarnis 
I have tried changing the usb cable but with no effect. The problem is the same **chip reset**  in the middle of flash. So I'm leaving the USB Cable issue out of the question now. How to capture this reset reason and listen over FTDI serial to USB adapter? I don't have the pinouts available for me to try the i**nternal USB-Serial/JTAG peripherals**.  Is there any other way to try it?

radimkarnis, 2024-06-24:
> Unfortunately, this module is attached to another host MCU by Uart connection

Is it possible to try flashing without this connection? Can you fully disable the other MCU while flashing? There is a possibility of the other MCU interfering with the serial connection. 

> it is Always rst:0x1 (POWERON) while on download mode, is it similar type definition?

No, there is a multitude of possible reset reasons, doesn't always have to be a POWERON reset. 

> FTDI DTR connected to GPIO9 and EN is left floating to get the DOWNLOAD BOOT MODE

EN left floating seems suspicious. Usually, it is connected to the FTDI RTS, which pulls EN low to reset the chip when needed. How do you enter the download mode, do you do it manually? Pulling EN low and then high triggers the POWERON reset. 

If you keep it floating (not really floating, there is an internal pull-up resistor keeping EN high when not used) during flashing, is any of your other HW interfacing with the EN pin?

MahadiHasantauhid, 2024-06-24:
The device comes with this STM32 host MCU connected via Uart. Unfortunate, it is not possible to free up the ESP32-C3. Even though this UART connection is done on pins (pins 18&19) while I'm trying to flash the code using pins ( 30&31). 

Actually, The EN Pin in connected to a switch with the ground using 10k resistor, I used the FTDI RTS pin and connected it, but that doesn't have any impact at all. Unless switch is pressed  the EN remain. As per schematics it is connected to 3.3v with 10k pull up resistor as you know. POWERON reset doesn't require anything for this module. Even if the button is not pressed I still get the POWERON reset. EN doesn't have any HW interfacing in this module. GPIO9 is connected to DTR pin to activate download mode. Even without DTR pin, I hooked GPIO9 pin to ground using switch to activate same download mode. `rst:0x1 (POWERON),boot:0x5 (DOWNLOAD(USB/UART0/1)) waiting for download ` . 

The message log mentioned in my first message keeps continuously pops up on serial terminal of Tera Term or Putty. Does that mean that the module keeps resetting? or is it normal a normal message Log?

radimkarnis, 2024-06-25:
> The message log mentioned in my first message keeps continuously pops up on serial terminal of Tera Term or Putty. Does that mean that the module keeps resetting? or is it normal a normal message Log?

That's is a normal execution log, seems like a pre-loaded app to print the module name and TX power - 
`ESP-ROM:esp32c3-api1-20210207` - ROM bootloader runs
`I (31) boot: ESP-IDF qa-test-v4.3.3-20220423 2nd stage bootloader` - 2nd stage bootloader runs (this is built by ESP-IDF)
`module_name:MINI-1` - the app runs

This is normal execution mode (not download mode) - the app is loaded from the SPI flash memory and executed.

> EN does have GPIO of STM32 interfacing in this module.

Is there a chance the STM32 is resetting the ESP? If the GPIO pulls the EN pin of the ESP to GND, it will reset.

I am sorry, but at this point I don't know how else to help you. I believe this is a HW issue that's causing your chip to reset. Having the ESP32-C3 being a part of a bigger device with other MCUs on it makes things even harder to debug. I advise you to study the [boot mode selection guide](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/advanced-topics/boot-mode-selection.html) and check if everything is ok and the EN pin is not getting asserted by the other MCU.

MahadiHasantauhid, 2024-06-27:
I have wiped clean the STM32 to check whether this is causing any issues or not. I think the reset reason is changed. 

`ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d1512
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x18c8
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2d6c
entry 0x403ce000
I (35) boot: ESP-IDF qa-test-v4.3.3-20220423 2nd stage bootloader
I (35) boot: compile time 07:45:53
I (35) boot: chip revision: 4
I (38) boot_comm: chip revision: 4, min. bootloader chip revision: 3
I (45) boot.esp32c3: SPI Speed      : 40MHz
I (50) boot.esp32c3: SPI Mode       : DIO
I (55) boot.esp32c3: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (91) boot:  2 nvs              WiFi data        01 02 00010000 0000e000
I (98) boot:  3 at_customize     unknown          40 00 0001e000 00042000
I (106) boot:  4 ota_0            OTA app          00 10 00060000 001d0000
I (113) boot:  5 ota_1            OTA app          00 11 00230000 001d0000
I (121) boot: End of partition table
I (125) boot: No factory image, trying OTA 0
E (130) esp_image: image at 0x60000 has invalid magic byte (nothing flashed here?)
E (138) boot: OTA app partition slot 0 is not bootable
E (144) esp_image: image at 0x230000 has invalid magic byte (nothing flashed here?)
E (152) boot: OTA app partition slot 1 is not bootable
E (158) boot: No bootable app partitions in the partition table`

MahadiHasantauhid, 2024-06-27:
@radimkarnis I flashed the module and got it connected to the wifi. I think you are right, somehow STM32 was causing this issue. I will have be sure by checking the code that was flashed into STM32. Thank you very much for your kind support. I will investigate further.

radimkarnis, 2024-06-28:
@MahadiHasantauhid Great, I am glad we could make some progress!

Ultimately - I think we can agree that this is not an esptool.py issue, but a hardware one. Do you agree to close this issue? If you find more information, you can always write it here for future generations to see.

MahadiHasantauhid, 2024-07-01:
I tried to flash  ESP-AWS-IOT example MQTT/tls_mutual_auth, and flash was successfull. But receiving the following issue 
`Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403808e2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x170c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (35) boot: ESP-IDF v5.1.4-dirty 2nd stage bootloader
I (35) boot: compile time Jul  1 2024 09:52:00
I (35) boot: chip revision: v0.4
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (43) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (77) boot:  1 nvs              WiFi data        01 02 0000f000 00009000
I (84) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (91) boot:  3 factory          factory app      00 00 00020000 00100000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00020020 vaddr=3c0b0020 size=28060h (163936) map
I (138) esp_image: segment 1: paddr=00048088 vaddr=3fc91400 size=02b04h ( 11012) load
I (141) esp_image: segment 2: paddr=0004ab94 vaddr=40380000 size=05484h ( 21636) load
I (148) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=a9a90h (694928) map
I (265) esp_image: segment 4: paddr=000f9ab8 vaddr=40385484 size=0bdb4h ( 48564) load
I (280) boot: Loaded app from partition at offset 0x20000
I (280) boot: Disabling RNG early entropy source...
I (292) cpu_start: Unicore app
I (292) cpu_start: Pro cpu up.
I (301) cpu_start: Pro cpu start user code
I (301) cpu_start: cpu freq: 160000000 Hz
I (301) cpu_start: Application information:
I (304) cpu_start: Project name:     tls_mutual_auth
I (309) cpu_start: App version:      202210.01-LTS-release-14-g40feb
I (316) cpu_start: Compile time:     Jul  1 2024 10:28:17
I (323) cpu_start: ELF file SHA256:  c7db97a3af42ce2c...
I (328) cpu_start: ESP-IDF:          v5.1.4-dirty
I (334) cpu_start: Min chip rev:     v0.3
I (339) cpu_start: Max chip rev:     v1.99
I (343) cpu_start: Chip rev:         v0.4
I (348) heap_init: Initializing. RAM available for dynamic allocation:
I (355) heap_init: At 3FC98B90 len 00027470 (157 KiB): DRAM
I (362) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (369) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (376) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (383) spi_flash: detected chip: generic
I (387) spi_flash: flash io: dio
W (391) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (405) sleep: Configure to isolate all GPIO pins in sleep state
I (411) sleep: Enable automatic switching of GPIO sleep configuration
I (418) app_start: Starting scheduler on CPU0
I (423) main_task: Started on CPU0
I (423) main_task: Calling app_main()
I (423) MQTT_EXAMPLE: [APP] Startup..
I (433) MQTT_EXAMPLE: [APP] Free memory: 279964 bytes
I (433) MQTT_EXAMPLE: [APP] IDF version: v5.1.4-dirty
I (443) example_connect: Start example_connect.
I (443) pp: pp rom version: 9387209
I (453) net80211: net80211 rom version: 9387209
I (463) wifi:wifi driver task: 3fca12b4, prio:23, stack:6656, core=0
I (463) wifi:wifi firmware version: 3ce09e5
I (463) wifi:wifi certification version: v7.0
I (473) wifi:config NVS flash: enabled
I (473) wifi:config nano formating: disabled
I (483) wifi:Init data frame dynamic rx buffer num: 32
I (483) wifi:Init static rx mgmt buffer num: 5
I (483) wifi:Init management short buffer num: 32
I (493) wifi:Init dynamic tx buffer num: 32
I (493) wifi:Init static tx FG buffer num: 2
I (503) wifi:Init static rx buffer size: 1600
I (503) wifi:Init static rx buffer num: 10
I (503) wifi:Init dynamic rx buffer num: 32
I (513) wifi_init: rx ba win: 6
I (513) wifi_init: tcpip mbox: 32
I (523) wifi_init: udp mbox: 6
I (523) wifi_init: tcp mbox: 6
I (523) wifi_init: tcp tx win: 5760
I (533) wifi_init: tcp rx win: 5760
I (533) wifi_init: tcp mss: 1440
I (543) wifi_init: WiFi IRAM OP enabled
I (543) wifi_init: WiFi RX IRAM OP enabled
I (553) phy_init: phy_�,Apr 3p��4,10:49:24
I BOD: Brownout detector was triggered`
I think at some point Brownout reset is triggered. Simple wifi station example program was just running fine previously. Still the STM32 is clean, no interference yet i guess. Do you see anything out of the ordinary from the log file? @radimkarnis

MahadiHasantauhid, 2024-07-01:
`E (230638) coreMQTT: sendBuffer: Unable to send packet: Network Error.
E (230648) coreMQTT: Transport send failed for DISCONNECT packet.
E (230658) coreMQTT: Sending MQTT DISCONNECT failed with status=MQTTSendFailed.` 
this is the latest issue after successfully flashing the ESP-AWS-IOT example code mqtt/tls_mutual_authentication...

closed 2024-07-08 by dobairoland

radimkarnis, 2024-07-08:
@MahadiHasantauhid I am sorry, but this is not an esptool issue anymore. I am not able to debug your application logs or solve hardware issues.

I advise you to focus on the brownout detector triggering. I believe your custom hardware has insufficient power supply or other power delivery issues, that's why you are seeing `Brownout detector was triggered`. 

You can also use our [free-of-charge schematic and PCB review service](https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review) to check the PCB and design.

For these reasons, we are now closing the issue.

MahadiHasantauhid, 2024-07-08:
@radimkarnis I would like to thank you for the support so far. You're right. The issues are related to hardware. Can't thank you enough for the support 🙏.