espressif/esptool #985: intelhex.HexRecordError: Hex file contains invalid record, when not using hex-files

bryghtlabs-richard, 2024-06-21:
### Operating System

Windows 11

### Esptool Version

v4.7.0

### Python Version

3.11.9

### Chip Description

esp32-s3-wroom-1 N16R8

### Device Description

Custom boardgame.

### Hardware Configuration

Strapping&Reset pins:
Reset - managed by ESP32
GPIO0: external 40k pull-up(shared strapping pin for another IC)
GPIO3: 10k pullup
GPIO45 Pull-down 0
GPIO46 Pull-down 0
All other pins are used

### How is Esptool Run

No, esptool from windows python pip. 

### Full Esptool Command Line that Was Run

python -m esptool -b 1000000 --before default_reset --after hard_reset --chip esp32s3 write_flash -e --flash_mode keep --flash_size 16MB --flash_freq 80m 0x0 .\ScriptFlashFiles\Firmware\ESP32\build\bootloader\bootloader.bin 0xc000 .\ScriptFlashFiles\Firmware\ESP32\build\partition_table\partition-table.bin 0xd000 .\ScriptFlashFiles\Firmware\ESP32\build\phy_init_data.bin 0xe000 .\ScriptFlashFiles\Esp32_Serial.bin 0xf000 .\ScriptFlashFiles\Firmware\ESP32\build\ota_data_initial.bin 0x2e0000 .\ScriptFlashFiles\Firmware\ESP32\blitzboot\build\blitz_boot.bin 0x370000 .\ScriptFlashFiles\Firmware\ESP32\blitzboot\build\blitz_boot.bin 0x400000 .\ScriptFlashFiles\Firmware\ESP32\build\blitz_app.bin

### Esptool Output

```shell
Traceback (most recent call last):
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\intelhex\__init__.py", line 115, in _decode_record
    bin = array('B', unhexlify(asbytes(s[1:])))
                               ^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\intelhex\compat.py", line 54, in asbytes
    return s.encode('latin1')
           ^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2030' in position 1: ordinal not in range(256)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\__main__.py", line 9, in <module>
    esptool._main()
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\__init__.py", line 1139, in _main
    main()
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\__init__.py", line 669, in main
    args = parser.parse_args(argv)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 1874, in parse_args
    args, argv = self.parse_known_args(args, namespace)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 1907, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 2110, in _parse_known_args
    positionals_end_index = consume_positionals(start_index)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 2087, in consume_positionals
    take_action(action, args)
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 1983, in take_action
    action(self, namespace, argument_values, option_string)
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 1246, in __call__
    subnamespace, arg_strings = parser.parse_known_args(arg_strings, None)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 1907, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 2131, in _parse_known_args
    stop_index = consume_positionals(start_index)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 2087, in consume_positionals
    take_action(action, args)
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\argparse.py", line 1983, in take_action
    action(self, namespace, argument_values, option_string)
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\__init__.py", line 1113, in __call__
    argfile = intel_hex_to_bin(argfile, address)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\esptool\bin_image.py", line 51, in intel_hex_to_bin
    ih.loadhex(file.name)
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\intelhex\__init__.py", line 212, in loadhex
    decode(s, line)
  File "C:\Users\Tester1\AppData\Local\Programs\Python\Python311\Lib\site-packages\intelhex\__init__.py", line 118, in _decode_record
    raise HexRecordError(line=line)
intelhex.HexRecordError: Hex file contains invalid record at line 1
```

### More Information

The command succeeds at least 9/10 times. It has never failed twice in a row. However, it has worked and failed using the same inputs with the same board.

### Other Steps to Reproduce

This error is unexpected, because we are not flashing any HEX files, only BIN files. I don't think it is hardware related because it seems to be a hex-file parsing error, but its strange that sometimes it fails and sometimes it works using the same input files.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

peterdragun, 2024-06-24:
Hi @bryghtlabs-richard, I am sorry for the issues. This is a known issue that was already solved in the master branch. Can you please try to install the latest dev release of esptool to verify that it fixes your problem?
You can install it using pip with the following command:
```
pip install esptool==4.8.dev4
```
Please let me know if that helps.

bryghtlabs-richard, 2024-07-01:
Have not seen this failure since updating, thanks!

closed 2024-07-01 by bryghtlabs-richard