espressif/esptool #992: Don't use self-hosted builders

floitsch, 2024-07-08:
### Is your feature request related to a problem?

The GitHub actions use a self-hosted builder which makes forking the repository harder.

### Describe the solution you'd like

You could run the build-steps on GitHub runners in a virtual machine.

Our implementation is here: https://github.com/toitlang/esptool/blob/b2d74c9958b1e6e858b655fbc1b6e28d3b3ca9be/.github/workflows/build.yml

The biggest disadvantage is that the building is much slower.

floitsch, 2024-07-23:
I just noticed that the linked build.yml script had an LGPL license. This is now changed to BSD-0: https://github.com/toitlang/esptool/blob/toitlang/.github/workflows/build.yml

FlorianHeigl, 2024-09-06:
it would make more sense if you'd title this "also allow running on github builders".
a project should be able to chose its infra as the project seems fit, and should be free to use the platforms' features as they seem fit.
(they should also document how to get it to a ci-buildable/forkable state to avoid shooting their own leg)

closed 2024-11-29 by espressif-bot