espressif/esptool #1000: `merge_bin` needs to validate the input file sizes

vshymanskyy, 2024-08-10:
### Is your feature request related to a problem?

If the file size exceeds the partition size, the `esptool` still finishes with success, presumably producing a broken firmware.

peterdragun, 2024-08-12:
Hi @vshymanskyy,
if I understand correctly you would like esptool to detect overlaps based on the partition table?

In the current implementation, esptool is detecting overlaps of binary files when running the merge_bin command. However esptool is not trying to understand the data during merging and does not have any information regarding the partition sizes, so I don't think this would be possible. Detecting such overlaps should be done on the upper level, in some tool that integrates esptool and parttool.

vshymanskyy, 2024-08-12:
I think `merge_bin` has everything it needs to validate the input. If the upper infrastructure implements this check, it effectively can merge the bin by itself, using esptool for this becomes redundant.