espressif/esptool #1001: Can I flash WiFi ssid and pw into default NVS?

hyansuper, 2024-08-20:


dobairoland, 2024-08-20:
You can flash anything (even random bytes) to anywhere in the flash. But you  (or the the upper application layer of ESP-IDF) have to know what goes and to where. Please consult the ESP-IDF documentation to find more about this. Esptool directly cannot help you with this. And we, esptool developers, are not able to answer this question.

closed 2024-08-20 by dobairoland

dobairoland, 2024-08-20:
This is probably a good starting point: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html