espressif/esptool #1004: Custom reset to reboot

boborjan2, 2024-09-01:
### Is your feature request related to a problem?

On boards that have custom setup and EN and IO0 is not wired as on "normal" boards, esptool already supports a custom reset strategy to enter bootloader mode. This works just fine. However the artermath reset ('--after') only supports limited options.

### Describe the solution you'd like

It would be great to be able to define custom reset for resetting the board just like in 'custom_reset_sequence'. This way resetting would be far more generic supporting boards with unusual setup. Existing code for parsing custom reset strategy could be reused.

radimkarnis, 2024-09-03:
Hello @boborjan2,

thank you for the feature request!

Please let me first understand your situation correctly. 

> resetting would be far more generic supporting boards with unusual setup

Defining a custom reset sequence is intended to help with variable development environments. Some combinations of OS, drivers, and HW can cause the timing requirements to change. The feature is not primarily implemented to support cases when "EN and IO0 is not wired as on "normal" boards". It is a response to the bootloader reset sequence failing in some cases - we had a few reports of this happening. On the other hand, there were never any reports about the --after reset needing any changes.

The --after reset is usually just a simple HIGH->LOW->HIGH assertion to the RST (`EN`) pin. Can you please describe how the "boards with unusual setup" could benefit from being able to change this? Are there any real cases when the RST (`EN`) is unavailable and the device has a more complicated reset circuitry?

This is just to understand the situation better so we can assess the feature. Thank you!

boborjan2, 2024-09-03:
Thanks for the response. In my example, defining a longer reset time would enough. Just as I needed this in the custom_reset_sequence. But even though I define my custom reset sequence, at the end (--after) esp_tool still simply calls HardReset that does not reset the board.

radimkarnis, 2024-09-04:
Understandable, could you please share more about how your board is an "unusual setup"? Do you know what the cause for needing a longer delay is?

boborjan2, 2024-09-04:
The reason for the long reset time is that EN is not wired to the debug connector. Instead, power-reset is used by an external circuitry. This way reset impulse must be long enough for the supply capacitors to discharge.

radimkarnis, 2024-09-23:
@boborjan2 thank you for clarification. It sounds reasonable, we will implement this, I will keep this ticket open until then.

closed 2024-10-21 by espressif-bot