espressif/esptool #1005: Unable to flash but build is successful

saimanohar1999, 2024-09-02:
### Operating System

ubuntu linux

### Esptool Version

4.8.dev5

### Python Version

python 3.10.12

### Chip Description

ESP32 S3

### Device Description

ESP32 S3 Dev kit

### Hardware Configuration

no hardware is connected

### How is Esptool Run

no IDE i am using Ubuntu with esp-idf tool

### Full Esptool Command Line that Was Run

idf.py -p /dev/ttyUSB0 flash

### Esptool Output

```shell
administrator@che-ll-caeteam01:~/400k/esp/esp-idf/examples/esp32_nat_router$ idf.py -p /dev/ttyUSB0 flash
Executing action: flash
Running ninja in directory /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build
Executing "ninja flash"...
[1/5] cd /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/esp-idf/espto...n /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/esp32_nat_router.bin
esp32_nat_router.bin binary size 0x100130 bytes. Smallest app partition is 0x12c000 bytes. 0x2bed0 bytes (15%) free.
[1/1] cd /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/bootloader/es...me/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/bootloader/bootloader.bin
Bootloader binary size 0x4a10 bytes. 0x35f0 bytes (42%) free.
[2/3] cd /home/administrator/400k/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D ID...r/build -P /home/administrator/400k/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 esp32_nat_router.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.dev5
Serial port /dev/ttyUSB0
Connecting....

A serial exception error occurred: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:66 (message):
  
  /home/administrator/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/home/administrator/400k/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed.

FAILED: CMakeFiles/flash /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/CMakeFiles/flash 
cd /home/administrator/400k/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/administrator/400k/esp/esp-idf -D "SERIAL_TOOL=/home/administrator/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/home/administrator/400k/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build -P /home/administrator/400k/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/log/idf_py_stderr_output_3627712 and /home/administrator/400k/esp/esp-idf/examples/esp32_nat_router/build/log/idf_py_stdout_output_3627712
```

### More Information

i tried multiple times still no use. even example code i am unable to flash.

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-09-03:
Hello @saimanohar1999,

are you selecting the correct port (`/dev/ttyUSB0`)? 

Esptool is a serial communication application, without a proper connection it cannot work. Please read the [troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html) for common causes of failing communication between the host and the ESP device.

saimanohar1999, 2024-09-04:
yes i am using the correct port /dev/ttyUSB0 i even verified it by connecting and disconnecting the target. its ttyUSB0

please help me solving this

radimkarnis, 2024-09-04:
Does your devkit have two USB ports? I believe you are connecting to the onboard USB-to-UART bridge (e.g. cp2102). This should work regardless of the ESP (it would work even if there was no ESP chip, because you are actually connecting to the USB bridge). That means this is not an issue with esptool or the ESP, but with the drivers, OS, or your hardware. All of these are really out of my reach and I can't help you.

1) Try changing the USB cable, try to get rid of any USB hubs. 
2) Run the `lsusb` command and see if you can get info about the device (post the results here).
3) Try connecting with another host machine.
4) Disconnect all of the GPIO pins of the devkit from other stuff.

closed 2024-09-16 by dobairoland