espressif/esptool #1006: Espressif's documentation is unclear

Joebeazelman, 2024-09-11:
### Operating System

NA

### Esptool Version

NA

### Python Version

NA

### Full Esptool Command Line that Was Run

NA

### Esptool Output

```shell
It''s unclear what Espressif defines as "ROM" and what it contains. Some vendors define ROM to include flash storage (Flash ROM), however, ROM traditionally means storage containing non-volatile, read-only, permanent or semi-permanent data programmed at the factory, which can't be modified without either replacement or reprogramming using specialized equipment.

Reading through the documentation, it claims the first stage boot loader is in "ROM". There's also evidence of this reading through the boot loader scripts containing "provides" for various runtime functions and Newlib. This is a bit odd, since such code is ordinarily part of the firmware rather than stored in ROM so it can be easily updated in response to a last-minute change, bug, or vulnerability by the vendor.

There's evidence also supporting that it's not "ROM". There are different linker scripts for different build scenarios. There's what appears to be a full and nano size Newlib definition, suggesting the code resides in flash and possibly shared across multiple apps. If indeed the code is stored "ROM" rather than Flash, then there would be no need to have different linker scripts, since the number function definitions has no impact on the code size.

So which one is it, ROM or Flash? What does it contain?
```

dobairoland, 2024-09-11:
> ROM traditionally means storage containing non-volatile, read-only, permanent or semi-permanent data programmed at the factory,
> Reading through the documentation, it claims the first stage boot loader is in "ROM".

Correct statements.

The 2nd stage bootloader is in the flash. That is the bootloader which is built and flashed together with the application and partition table when you try a very simple ESP-IDF example.

closed 2024-09-11 by dobairoland

Joebeazelman, 2024-09-12:
Are there any libraries in the "ROM" where the 1st stage boot loader is located?

dobairoland, 2024-09-12:
https://esp32.com/viewtopic.php?f=13&t=41784