espressif/esptool #1014: pip install esptool -U --prefer-binary installs 4.8.0

matthuisman, 2024-09-25:
### Operating System

Windows

### Esptool Version

4.8.0

### Python Version

3.10

### Other Steps to Reproduce

`pip install esptool -U --prefer-binary`
will install the "faulty" 4.8.0 whl as I suspected in comment https://github.com/espressif/esptool/pull/1011#issuecomment-2369088764

I'd recommend trying to remove that old whl.
If thats not possible, then yanking the 4.8.0 alltogether

matthuisman, 2024-09-25:
fyi @radimkarnis  :)

radimkarnis, 2024-09-26:
@matthuisman we will look into it, thank you!

radimkarnis, 2024-09-26:
v4.8.0 wheel deleted. Thanks again for your help!

closed 2024-09-26 by radimkarnis

matthuisman, 2024-09-26:
thats great :) Thank you