espressif/esptool #1015: Clarify supported SPI flash for use with --spi-connection

nebkat, 2024-09-27:
### Is your feature request related to a problem?

I have been trying to no avail to get a variety of NAND flash chips to work with `esptool read_flash`. My understanding is that NAND flash is not supported in ROM due to its complexity?

The [Overriding SPI Flash Connections](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/advanced-options.html#overriding-spi-flash-connections) documentation does not sufficiently explain the requirements for interfacing with an SPI flash chip.

### Describe the solution you'd like

Could we get clear documentation on what type of chips are supported, as many are advertised as "SPI flash".

radimkarnis, 2024-09-30:
Hello @nebkat,

NAND flash chips are not supported, the ROM doesn't implement the needed driver. You need a NOR flash chip capable of at least DIO mode.

We don't keep a list of all supported chips. You can see the [flashrom source code](https://github.com/flashrom/flashrom/blob/master/include/flashchips.h) to see the most common flash chips. Other extra modes, such as OPI flash and 32-bit addressing are described [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#optional-features-for-flash), but that doesn't include NAND flash chips.

I will add a note to esptool docs to explain this.

> Could a custom flasher stub be made that is able to communicate with more types of flash?

It could, but that would only help with flashing and reading the data with esptool. If you wanted to use the flash memory in your app, you would have to re-implement the driver in your code, because the ROM, ESP-IDF and Arduino don't support it. There might also be other reasons why it's not supported I am not aware of (e.g. HW limitations).

nebkat, 2024-09-30:
Hey @radimkarnis,

Thank you for confirming my suspicions - yes if a note could be added I hope it will save someone else some confusion!

> It could, but that would only help with flashing and reading the data with esptool.

This is what I would be most interested in. We use NAND flash in all of our products for user data storage, would be amazing if we could pull data using an esptool-like interface. Perhaps something based on https://github.com/espressif/idf-extra-components/tree/master/spi_nand_flash and `esptool load_ram`. I will look in to it!

closed 2024-10-03 by espressif-bot