espressif/esptool #1016: Cannot upload sketch to NodeMCU ESP-32S V1.1

nicolasgomez1, 2024-10-06:
### Operating System

Windows 11 23H2 (22631.4249)

### Esptool Version

-

### Python Version

4.6

### Chip Description

NodeMCU ESP-32S V1.1

### How is Esptool Run

Arduino IDE 2.3.3

### Full Esptool Command Line that Was Run

"C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6/esptool.exe" --chip esp32 --port "COM4" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000 "C:\Users\nicoh\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bootloader.bin" 0x8000 "C:\Users\nicoh\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.partitions.bin" 0xe000 "C:\Users\nicoh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\nicoh\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bin" 

### Esptool Output

```shell
Sketch uses 1065865 bytes (81%) of program storage space. Maximum is 1310720 bytes.
Global variables use 48052 bytes (14%) of dynamic memory, leaving 279628 bytes for local variables. Maximum is 327680 bytes.
"C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6/esptool.exe" --chip esp32 --port "COM4" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bootloader.bin" 0x8000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.partitions.bin" 0xe000 "C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bin" 
esptool.py v4.6
Serial port COM4
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

### More Information

Sorry if I filled out the issue data wrong, I'm new to this... It turns out I bought a NodeMCU ESP-32S V1.1 recently. And I can't upload the sketches like I did with my ESP8266. The only way I found was to press the 100 button (the board has another one that says EN, I don't press that one)

First I tried with the drivers: CP210x_Universal_Windows_Driver. Then with: CP210x_VCP_Windows. But nothing works. I see that some say that you can modify the esptool.py, but I don't have that file, only esptool.exe

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2024-10-07:
Hi @nicolasgomez1. The troubleshooting guide contains a section about the "wrong boot mode": https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#wrong-boot-mode-detected-0xxx-the-chip-needs-to-be-in-download-mode. That section links the manual bootloader section: https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader. I guess you found this.

I think you also guessed that the issue is with the Windows driver. We introduced a workaround (feature) to be able to specify custom reset sequences and work around issues with the Windows driver. You are on a year and a half old esptool version. Let me check if this is available....

dobairoland, 2024-10-07:
Yes, custom reset sequences are available since v4.5:  https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html#custom-reset-sequence

You can change the reset sequence by the configuration file in case you have only an `esptool.exe` and can achieve the same than hacking into esptool and modify it there.

nicolasgomez1, 2024-10-07:
@radimkarnis

If the button says IO0, my apologies, the PCB etching typography is not very clear. I updated to version 4.8 of esptool and ran the flash_id command.

the output was
```
esptool.py v4.8.1
Serial port COM4
Connecting...............
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c0:5d:89:af:9c:b8
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
Flash voltage set by a strapping pin to 3.3V
Hard resetting via RTS pin...
```

radimkarnis, 2024-10-08:
@nicolasgomez1 

you can see your chip now successfully connects! So the latest version of esptool solves your issues.

It seems like the latest version used in Arduino is v4.6 - so you have two options:
1) Either you run the command manually with the latest esptool version (`esptool.py --chip esp32 --port "COM4" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bootloader.bin" 0x8000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.partitions.bin" 0xe000 "C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bin`)
1) Or you manually replace the `esptool.exe` binary in the Arduino tools installation directory (`/tools/esptool`) with one downloaded from [the last release](https://github.com/espressif/esptool/releases/tag/v4.8.1). This is more user friendly in the long run, but it's not guaranteed to work -  the Arduino team hasn't tested the latest release of esptool yet.

nicolasgomez1, 2024-10-08:
> @nicolasgomez1
> 
> you can see your chip now successfully connects! So the latest version of esptool solves your issues.
> 
> It seems like the latest version used in Arduino is v4.6 - so you have two options:
> 
> 1. Either you run the command manually with the latest esptool version (`esptool.py --chip esp32 --port "COM4" --baud 115200  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bootloader.bin" 0x8000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.partitions.bin" 0xe000 "C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.5/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\XXX\AppData\Local\Temp\arduino\sketches\A11E3EF2B4A234F97DA8C0291DCB4111/ESP32Indoor.ino.bin`)
> 2. Or you manually replace the `esptool.exe` binary in the Arduino tools installation directory (`/tools/esptool`) with one downloaded from [the last release](https://github.com/espressif/esptool/releases/tag/v4.8.1). This is more user friendly in the long run, but it's not guaranteed to work -  the Arduino team hasn't tested the latest release of esptool yet.

No no, version 4.8.1 doesn't solve the problem. It is still necessary to press the IO0 button...

As for the Arduino IDE, I found a bit of a dirty solution. Replace the line
```
tools.esptool_py.path={runtime.tools.esptool_py.path}
```
With
```
tools.esptool_py.path={runtime.tools.esptool_py-4.8.path}
```
In the platform.text file.

As for esptool, in the links shared by @dobairoland it seems that it is possible to use a configuration file. Is that possible with the binary version of the tool?

radimkarnis, 2024-10-09:
> No no, version 4.8.1 doesn't solve the problem. It is still necessary to press the IO0 button...

Well, that wasn't apparent from your log. In that case, you will have to resort to trying to change the reset sequence.

> As for the Arduino IDE, I found a bit of a dirty solution. Replace the line

If v4.8.1 doesn't solve the issue, replacing esptool in Arduino won't do anything.

> Is that possible with the binary version of the tool?

Yes, the config file option works even with the binary version (`esptool.exe`).

nicolasgomez1, 2024-10-14:
> Well, that wasn't apparent from your log. In that case, you will have to resort to trying to change the reset sequence.
>
>
> If v4.8.1 doesn't solve the issue, replacing esptool in Arduino won't do [...]

Hmm, sorry to insist, but how do I do it? I know I have to add a debound, but in the parameters that are listed when sending the -h argument, there is nothing about it...

nicolasgomez1, 2024-10-14:
> Well, that wasn't apparent from your log. In that case, you will have to resort to trying to change the reset sequence.
>
>
> If v4.8.1 doesn't solve the issue, replacing esptool in Arduino won't do [...]

excuse me for my ignorance, but I don't know how to do it. I pass the -help argument to esptool.exe and in the list of commands it returns, I don't see any that indicate that it can be used to pass a different sequence or pass the path of a file with the new sequence. [I see that it can be done with the .py version](https://github.com/espressif/esptool/issues/706) but I don't understand how to do it with the binary.

radimkarnis, 2024-10-14:
You need to create a config file in the correct location. There is no difference in how to do this between the `.py` and `.exe` versions of esptool. Detailed instructions are [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html) - these describe where to create the config file, its format, and how to craft a custom reset sequence.

radimkarnis, 2024-10-29:
This seems to be a hardware issue, or OS drivers issue. There's probably not much we can do on the software side. For this reason and for inactivity, I am closing this issue.

closed 2024-10-29 by radimkarnis