espressif/esptool #1019: MD5 of file does not match data in flash!

ljm996054544, 2024-10-11:
### Operating System

win Vscode+IDF_V5.1.4

### Esptool Version

v4.7.0

### Python Version

Python 3.11.2

### Chip Description

ESP35S3

### Hardware Configuration

nothing

### How is Esptool Run

win Vscode+IDF

### Full Esptool Command Line that Was Run

d:\APP\Espressif\idf_510\idf_tools\python_env\idf5.1_py3.11_env\Scripts\python.exe d:\APP\Espressif\idf_510\v5.1\esp-idf\components\esptool_py\esptool\esptool.py -p COM18 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 simple_ota.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin

### Esptool Output

```shell
esptool.py v4.7.0
Serial port COM18
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 34:85:18:98:3e:70
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000f2fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Compressed 20960 bytes to 13316...
Wrote 20960 bytes (13316 compressed) at 0x00000000 in 0.6 seconds (effective 296.3 kbit/s)...
File  md5: f88272169361160868e505a3542a7671
Flash md5: 516460e381627df19e0d1c86f0bcc013
MD5 of 0xFF is 516460e381627df19e0d1c86f0bcc013

A fatal error occurred: MD5 of file does not match data in flash!
```

### More Information

Burning ESP32c3 is normal, but ESP32s3 cannot be burned.
I will merge the three generated bin files and download them using the esphome tool（ https://web.esphome.io/ ）It can be downloaded normally and the chip is running normally

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

ljm996054544, 2024-10-11:
The idf5.1.0 previously installed on another computer was able to burn the same S3 development board normally. Now I have tried installing multiple versions of idf, but none of them can burn this ESP32S3 development board
Use https://web.esphome.io/ and flash_rownload_tool_3.9.7.exe tool can burn, which should indicate that my hardware is functioning properly

ljm996054544, 2024-10-14:
I have installed IDF4.4 and can burn ESP32S3 normally, but my project requires IDF5.1

ljm996054544, 2024-10-14:
Copy esptool.by from IDF4.4 to idf531, and idf531 will be able to burn esp32s3 normally. However, I don't know the specific reason for this. Currently, it can be used normally, and I'm not sure if there are any hidden dangers
[image]

ljm996054544, 2024-10-14:
> esptool.by

I was wrong. Replacing the esptool.by file can allow esp32s3 to burn normally, but it can also cause strange compilation issues

Dzarda7, 2024-10-14:
Hi @ljm996054544 can you please provide the images? I will try to reproduce it.

ljm996054544, 2024-10-18:
> Hi @ljm996054544 can you please provide the images? I will try to reproduce it.

[images.zip](https://github.com/user-attachments/files/17427457/images.zip)
I am using the hello_world demo

Dzarda7, 2024-10-21:
Thanks for sending the images. I was not able to reproduce the issue unfortunately. I see from the log that you tried v4.7.0 and v4.8.1 and neither worked for you.

Could you please try [installing an older version of esptool](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/installation.html) manually and see if e.g. `v4.6.2` work for you? The docs contain instructions on even how to change the esptool version in the context of ESP-IDF.

Sending the log with `--trace` option enabled in esptool.py would also be helpful.

You can also try to flash the images with `--no-compress` option and see what it does (this might be workaround but not the solution).

Jason2866, 2024-10-23:
The reason is the difference in flash size. Your flash command uses 4MB and the connected device has 8MB flash. To avoid such issues build the firmware images for 8MB.

ljm996054544, 2024-10-28:
> 原因是 Flash 大小的差异。您的 flash 命令使用 4MB，连接的设备使用 8MB flash。为避免此类问题，请构建 8MB 的固件映像。

I am using ESP32S3-N16R8, and I have set Flash to 2/4/8/16, but none of them work

ljm996054544, 2024-10-28:
> Thanks for sending the images. I was not able to reproduce the issue unfortunately. I see from the log that you tried v4.7.0 and v4.8.1 and neither worked for you.
>
> Could you please try [installing [...]

I manually installed esptool and tested one version after another. (Master) v4.6.2 can be used normally, but versions greater than 4.6.2 will report errors

Dzarda7, 2024-10-29:
@ljm996054544 okay, thanks for trying this, can you please send the logs with `--trace` option, one from v4.6.2 and one from newer (which does not work). I will take a look at it. Also, did you try the `--no-compress` option?

dobairoland, 2024-11-15:
I'm closing this for inactivity and because we cannot reproduce it. If you would be willing to provide more information then we will re-open it.