espressif/esptool #1020: Can't connect to esp8266

radio-satellites, 2024-10-12:
### Operating System

Windows/Fedora Linux

### Esptool Version

esptool.py v4.8.1 4.8.1

### Python Version

Python 3.12.6

### Chip Description

ESP8266

### Device Description

My own PCB. I've double checked all strapping pins, they seem fine. 

### Hardware Configuration

GPIO16 - SDN
GPIO4 - NIRQ
GPIO15 - NSEL
GPIO14 - SCK
GPIO13 - MOSI
GPIO12 - MISO

All of these are connected to an Si4432.

### How is Esptool Run

Terminal, arduino IDE

### Full Esptool Command Line that Was Run

python esptool.py --chip esp8266 --port COM9 --trace --baud 78400 --before default_reset --after hard_reset write_flash 0x0 C:\Users\sasha\AppData\Local\Temp\arduino_build_160028/sketch_oct10a.ino.bin

### Esptool Output

```shell
esptool.py v4.8.1
Serial port COM9
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.117 No serial data received.
.TRACE +0.091 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.125 No serial data received.
.TRACE +0.092 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
[... 505 lines cut ...]
    f9ead92c857128d1 aa386238a298a55e | ...,.q(..8b8...^
    59615e59a298a55e 596160796038a298 | Ya^Y...^Ya`y`8..
    a55e596160796038 6160796020a26a81 | .^Ya`y`8a`y` .j.
    6238a298a55e5961 6079603861607960 | b8...^Ya`y`8a`y`
    20a26a8162386160 7960206079603861 |  .j.b8a`y` `y`8a
    6079602062386220 a26a81a298a55e59 | `y` b8b .j....^Y
    a298a55e59616079 6038a298a55e5961 | ...^Ya`y`8...^Ya
    6079603861607960 20a26a816238a298 | `y`8a`y` .j.b8..
    a55e596160796038 6160             | .^Ya`y`8a`
TRACE +0.000 Remaining data in serial buffer:
    796020a26a816238 6160796020a26a81 | y` .j.b8a`y` .j.
    6220a26a81a2c06a 744e386220a2     | b .j...jtN8b .
.

A fatal error occurred: Failed to connect to ESP8266: Invalid head of packet (0x20): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### More Information

The boot mode is (1,6) - is this a problem? It appears that the command receives data, but it doesn't seem to process it. 

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

Honza0297, 2024-10-14:
Hello @radio-satellites!

Most of the output is `esptool` trying to sync with the board. Synchronization happens a few times, but is lost very quickly. Sometimes, the board enters FW download mode (boot mode (1,6)), sometimes, the board starts to output invalid data (`.j.b .j..8b8....` etc).  Your board is probably not able to stay in the FW download mode.  

This is almost certainly not a SW issue, but rather a HW one. 

First of all, do you have the [required pins](https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#required-pins) in correct states when booting? 

* *GPIO15 pulled-down* - I see you are using this pin, so please pay extra attention to this. 
* GPIO2 pulled-up

If yes, try to open serial monitor (e.g. python -m esp_idf_monitor -b 78400) with 

* GPIO pull down. This should get the board into FW download mode and some info might be printed out in the monitor.
* GPIO0 pulled up. This should enter normal boot and application might write something to the serial port. If the text will be gibberish, try other baudrates (9600, 115200...)

radio-satellites, 2024-10-15:
Hi @Honza0297!

Thanks for your reply! I just noticed it now. I'm not sure why GitHub isn't sending me emails anymore...

Yes, both GPIO15 and GPIO2 are pulled LOW, or at least both have pull-down resistors placed on the PCB. I did check with a multimeter, and for some reason GPIO15 reads positive (although it *is* connected to a 10k pull-down?). I connected it to ground via a jumper and it still doesn't want to flash, unfortunately. GPIO15 is used as NSEL for the Si4432 SPI interface (which should be LOW anyways). 

Yes, the GPIO0 modes are as expected. GPIO0 to GND outputs the expected flash state but pulled up boots into the regular default firmware.

Honza0297, 2024-10-16:
Thanks for your answer! 

> ...both GPIO15 and GPIO2 are pulled LOW... 

GPIO02 should be pulled **UP** (see the table here: https://docs.espressif.com/projects/esptool/en/latest/esp8266/advanced-topics/boot-mode-selection.html#required-pins ). If it is pulled LOW, it may be the culprit we are looking for.

radio-satellites, 2024-10-18:
@Honza0297 That was a typo on my part. I meant GPIO0. I'll double check GPIO2 in a sec...

radio-satellites, 2024-10-18:
Here's the schematic for the ESP8266 section, by the way
[image]

radio-satellites, 2024-10-18:
Not sure what happened... but...

[image]
 Thank you so much @Honza0297!

radio-satellites, 2024-10-19:
Guess I'll have to reopen...

Worked fine until all of a sudden:

```esptool.py v3.0
Serial port COM9
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.111 Timed out waiting for packet header
.TRACE +0.102 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.114 Timed out waiting for packet header
.TRACE +0.104 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
.TRACE +0.105 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.112 Timed out waiting for packet header
.TRACE +0.104 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.115 Timed out waiting for packet header
.TRACE +1.905 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.113 Timed out waiting for packet header
_TRACE +0.103 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.003 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.002 Read 1 bytes: c0
TRACE +0.004 Read 1 bytes: 01
[... 49 lines cut ...]
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.116 Timed out waiting for packet header
.TRACE +1.897 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.110 Timed out waiting for packet header
_TRACE +0.102 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.002 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.003 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 01000200000000000105c0
TRACE +0.001 Received full packet: 01000200000000000105

A fatal esptool.py error occurred: Invalid (unsupported) command 0x8

Honza0297, 2024-10-19:
Let's debug further. Can you desolder the Si442 temporarily? I see you stated:

> GPIO15 reads positive (although it is connected to a 10k pull-down?).

which may indicate an issue on the Si442 side.

radio-satellites, 2024-10-21:
Nope, it looks about the same. 

I'm wondering at this point if it's the reliability of the UART connection - I've gotten it to flash *sometimes*, but other times it just inexplicably fails.

radio-satellites, 2024-10-21:
UART is connected via 2.54mm headers into jumper wires, which I then plug into a USB to UART converter.

Honza0297, 2024-10-21:
You can try to switch cables to rule out possible issues with the wires, but generally speaking, jumper wires are good enough for UART communication, if no other issue is in play.  
 
 I am still wondering how HIGH value got on GPIOI15 before you shorted it directly to GND.  Please let me know if you are able to flash the ESP8266 without Si4432 soldered.

radio-satellites, 2024-10-22:
Yes, the chip flashes equally sometimes with and without the Si4432. 

Sometimes, changing jumper wires helps, but not always.

radio-satellites, 2024-10-22:
Another debug output from a failed flash:

```
esptool.py v3.0
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +1.804 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
_TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
_TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
[... 569 lines cut ...]
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
_TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
_TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
_

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header

Honza0297, 2024-10-22:
If all the pins are in correct states, this is most likely a HW issue, not an issue with esptool.
 
You can send your PCB design for free-of-charge review: https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review , where our engineers will check if the PCB design is correct.  

We are closing the issue for now.

radimkarnis, 2024-10-22:
@radio-satellites esptool requires a working serial connection to operate. Without it, there is not much we can do on the software side. Please try to inspect your serial chain again (e.g. use `miniterm` or `esp-idf-monitor` to open a console and see what output (if any) you get in certain cases).

We'll close this in the meantime. Feel free to reopen if you get more information.

closed 2024-10-22 by radimkarnis

radio-satellites, 2024-10-22:
@radimkarnis Yes, UART is definitely working and reliable. I'm consistently getting an output every time which indicates the boot mode and cause when GPIO0 is set to LOW and RST is triggered.

radio-satellites, 2024-11-01:
I finally found the issue. It is incredibly simple: sometimes, the GPS (using UART0) would not shutdown before flashing, causing the data from esptool to not reach the MCU. That's it!

jkubasa, 2025-05-01:
How are shutdown GPS?