espressif/esptool #1021: 'ESPLoader' object has no attribute 'CHIP_DETECT_MAGIC_VALUE'

brentpicasso, 2024-10-23:
### Operating System

Linux

### Esptool Version

4.8.1

### Python Version

3.11.9

### Chip Description

ESP32-S3

### Device Description

ESP32-S3 connected to host via USB

### Hardware Configuration

n/a

### How is Esptool Run

via python code

### Full Esptool Command Line that Was Run

via python code

### Esptool Output

```shell
Custom python code is crashing when detecting chip. 

AttributeError: 'ESPLoader' object has no attribute 'CHIP_DETECT_MAGIC_VALUE'
```

It appears there is a bug in the ESPLoader where the CHIP_DETECT_MAGIC_VALUE is not defined, in two places:

https://github.com/espressif/esptool/blob/master/esptool/loader.py#L760
https://github.com/espressif/esptool/blob/master/esptool/loader.py#L763

This is my code snippet:
```
                for try_port in available_ports:
                    Logger.info(f"Trying to detect bootloader on port {try_port}")
                    try:
                        esp = esptool.ESPLoader(port=try_port.device, baud=cls.INITIAL_BAUDRATE)
                        esp.connect()
                        chip_desc = esp.get_chip_description()
                        Logger.info(f"Detected bootloader on port {try_port} : {chip_desc}")
                        port = try_port
                        esp = esp.run_stub()
                        with open(filename, 'rb') as firmware:
                            firmware_data = firmware.read()
                            
                        esp.flash_data(firmware_data, progress_callback=_progress_callback)                            

                    except esptool.FatalError:
                        Logger.info(f"Bootloader not found on {try_port}")
```
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2024-10-23:
Hello @brentpicasso,
please see the docstring of the `ESPLoader` class:

``` 
    Don't instantiate this base class directly, either instantiate a subclass or
    call cmds.detect_chip() which will interrogate the chip and return the
    appropriate subclass instance. You can also use a context manager as
    "with detect_chip() as esp:" to ensure the serial port is closed when done.
```

Initiating the `esp` object as `esp = esptool.ESPLoader(port=try_port.device, baud=cls.INITIAL_BAUDRATE)` is wrong. Either use a specific ROM class (e.g. `ESP32C5ROM`) or interrogate the chip first.

brentpicasso, 2024-10-23:
thanks for the tip, I made further progress. :+1:

closed 2024-10-23 by brentpicasso