espressif/esptool #1026: CP2102 / USB-TO-UART chips customization functionality.

cederom, 2024-10-27:
### Is your feature request related to a problem?

* Many different devkits have the same VID:PID, Name, SN identificators.
* Build test farms (i.e. NuttX RTOS) that have many devkits connected via USB-TO-SERIAL adapters may not use boards with the same identification strings because often serial port device (i.e. `/dev/cuaU*`) numbers are assigned randomly on attach. Problem is even worse then 16 port USB HUBs are used with dynamic port power cycling.

### Describe the solution you'd like

* Ability to customize USB-TO-SERIAL chips (i.e. VID:PID, Name, SN) built directly into the esptool.
* It would enable easily identification of various develkits using the same USB-TO-UART chips based on unique VID:PID, Name, and SN and help in build/runtime testing automation.
* Add option to `esptool.py --port-filter` to also filter based on SN.

### Describe alternatives you've considered

* I found two reference utilitities that can customize CP2102 chip:
  * C: https://github.com/DiUS/cp210x-cfg
  * Python: https://github.com/cederom/cp210x-program
* `cp210x-cfg` works fine and allows Name and SN change to CP2102 on the ESP32-DevKitC several times.

### Additional context

```
% uname -a
FreeBSD octagon 13.3-RELEASE-p7 FreeBSD 13.3-RELEASE-p7 GENERIC amd64

% ./cp210x-cfg
ID 10c4:ea60 @ bus 000, dev 007: CP2102 USB to UART Bridge Controller
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: CP2102 USB to UART Bridge Controller
Serial: 0001

% dd if=/dev/random count=1M | md5
1048576+0 records in
1048576+0 records out
76f24aad88225fedf7260bb4f54c525d
536870912 bytes transferred in 3.788303 secs (141718053 bytes/sec)

% ./cp210x-cfg -h
Syntax:
cp210x-cfg [-h ] |
           [-m vid:pid] [-d bus:dev]
           [ -l | [-V vid] [-P pid] [-F flush] [-M mode] [-N name] [-S serial]]

  -h            This help
  -m vid:pid    Find and use first device with vid:pid
  -d bus:dev    Find and use device at bus:dev
  -l            List all CP210x devices connected
  -V vid        Program the given Vendor ID
  -P pid        Program the given Product ID
  -F flush      Program the given buffer flush bitmap (CP2105 only)
  -M mode       Program the given SCI/ECI mode (CP2105 only)
  -N name       Program the given product name string
  -S serial     Program the given serial string

Unless the -d option is used, the first found CP210x device is used.
If no programming options are used, the current values are printed.

% ./cp210x-cfg -S 76f24aad88225fedf7260bb4f54c525d
ID 10c4:ea60 @ bus 000, dev 007: CP2102 USB to UART Bridge Controller
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: CP2102 USB to UART Bridge Controller
Serial: 76f24aad88225fedf7260bb4f54c525d

% ./cp210x-cfg
ID 10c4:ea60 @ bus 000, dev 007: CP2102 USB to UART Bridge Controller
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: CP2102 USB to UART Bridge Controller
Serial: 76f24aad88225fedf7260bb4f54c525d

% ./cp210x-cfg -N ESP32DEVKITC-CDR-76f24aad88225fedf7260bb4f54c525d
ID 10c4:ea60 @ bus 000, dev 007: CP2102 USB to UART Bridge Controller
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: ESP32DEVKITC-CDR-76f24aad88225fedf7260bb4f54c525d
Serial: ESP32DEVKITC-CDR-76f24aad88225fedf7260bb4f54c525d

% ./cp210x-cfg -N ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
ID 10c4:ea60 @ bus 000, dev 007:
ESP32DEVKITC-CDR-76f24aad88225fedf7260bb4f54c525d
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
Serial: ESP32DEVKITC-CDR-76f24aad88225fedf7260bb4f54c525d

% ./cp210x-cfg
ID 10c4:ea60 @ bus 000, dev 007:
ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
Serial: ESP32DEVKITC-CDR-76f24aad88225fedf7260bb4f54c525d

% ./cp210x-cfg -S ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
ID 10c4:ea60 @ bus 000, dev 007:
ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
Serial: ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d

% ./cp210x-cfg
ID 10c4:ea60 @ bus 000, dev 007:
ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
Model: CP2102
Vendor ID: 10c4
Product ID: ea60
Name: ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
Serial: ESP32DEVKITC-CeDeROM-76f24aad88225fedf7260bb4f54c525d
```

radimkarnis, 2024-10-29:
Hi @cederom,
Thank you for the suggestion!

This is an interesting feature, but completely out of the scope of esptool.

Esptool is a utility for working with chips from the ESP family. It is agnostic to DevKit configuration and components, such as USB-to-UART bridges. What's more, this would only work for Silicon Labs cp210x chips. What about FTDI or CH340 chips? 

The idea to add a serial number filter option sounds good, we'll most likely include that, but the logic to program cp210x chips has to stay in a separate package.

zsteva, 2024-11-17:
simple use udev rules (at least on linux) to make uniq symlink based on serial number of chip chip.