espressif/esptool #1048: burn_key operation gives Burn BLOCK_KEY0 (['BLOCK4']) was not successful

Daphne-HU, 2025-01-09:
### Operating System

Xubuntu 20.04 (under WSL)

### Esptool Version

4.7.0

### Python Version

3.8.10

### Chip Description

esp32-C3-mini-1

### Device Description

Chip is on a custom PCB

### Hardware Configuration

When relevant i can provide pin configurations

### How is Esptool Run

Visual studio code

### Full Esptool Command Line that Was Run

espefuse.py --debug burn_key BLOCK_KEY0 flash_encryption_key.bin XTS_AES_128_KEY

### Esptool Output

```shell
vscode@4d6257b198cd:/workspaces/BCE-Monitor$ espefuse.py --debug burn_key BLOCK_KEY0 flash_encryption_key.bin XTS_AES_128_KEY
espefuse.py v4.7.0
Connecting...
Detecting chip type... ESP32-C3
BLOCK0          (                ) [0 ] read_regs: 00000000 00000000 00000000 00000000 80000000 00000000
MAC_SPI_8M_0    (BLOCK1          ) [1 ] read_regs: <data>
BLOCK_SYS_DATA  (BLOCK2          ) [2 ] read_regs: <data>
BLOCK_USR_DATA  (BLOCK3          ) [3 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY0      (BLOCK4          ) [4 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY1      (BLOCK5          ) [5 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY2      (BLOCK6          ) [6 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY3      (BLOCK7          ) [7 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY4      (BLOCK8          ) [8 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY5      (BLOCK9          ) [9 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_SYS_DATA2 (BLOCK10         ) [10] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

BLOCK0          (                ) [0 ] err__regs: 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE_RD_RS_ERR0_REG        0x00000000
EFUSE_RD_RS_ERR1_REG        0x00000000

=== Run "burn_key" command ===
Burn keys to blocks:
 - BLOCK_KEY0 -> [<my key>]
        Reversing byte order for AES-XTS hardware peripheral
        'KEY_PURPOSE_0': 'USER' -> 'XTS_AES_128_KEY'.
        Disabling write to 'KEY_PURPOSE_0'.
        Disabling read to key block
        Disabling write to key block

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[00] BLOCK0               is not empty
        (written ): 0x000000008000000000000000000000000000000000000000
        (to write): 0x<key>
        (coding scheme = NONE)
[01] MAC_SPI_8M_0         nothing to burn
[02] BLOCK_SYS_DATA       nothing to burn
[03] BLOCK_USR_DATA       nothing to burn
[04] BLOCK_KEY0           is empty, will burn the new value
[05] BLOCK_KEY1           nothing to burn
[06] BLOCK_KEY2           nothing to burn
[07] BLOCK_KEY3           nothing to burn
[08] BLOCK_KEY4           nothing to burn
[09] BLOCK_KEY5           nothing to burn
[10] BLOCK_SYS_DATA2      nothing to burn
. 
This is an irreversible operation!
Type 'BURN' (all capitals) to continue.
BURN
BLOCK_KEY0      (BLOCK4          ) [4 ] to_write: <my key>
Write data to BLOCK4
Addr 0x60008800, data=0x..
Addr 0x60008804, data=0x..
Addr 0x60008808, data=0x..
Addr 0x6000880c, data=0x..
Addr 0x60008810, data=0x..
Addr 0x60008814, data=0x..
Addr 0x60008818, data=0x..
Addr 0x6000881c, data=0x..
Addr 0x60008820, data=0x..
Addr 0x60008824, data=0x..
Addr 0x60008828, data=0x..
BLOCK_KEY0      (BLOCK4          ) [4 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY0      (BLOCK4          ) [4 ] Expected: <my key>
BLOCK_KEY0      (BLOCK4          ) [4 ] Real    : 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

A fatal error occurred: Burn BLOCK_KEY0 (['BLOCK4']) was not successful
```

### More Information

I followed the documentation as noted here, under Using Host Generated Key:
https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32c3/security/flash-encryption.html

When I run "espefuse.py summary," it displays what appears to be the basic fabric configuration settings.
Nothing is changed before trying to burn the flash key.

### Other Steps to Reproduce

-

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

KonstantinKondrashov, 2025-01-10:
Hi @Daphne-HU!
The command is correct. If nothing is changed then the problem is likely hardware-related. You can try programming it into a different BLOCK_KEY.
Possible hardware-related issues:
- ESD (Electrostatic Discharge): Ensure proper handling and grounding to prevent damage.
- Weak Power Supply: Verify the power supply's stability and ensure it meets the ESP32-C3's voltage and current requirements.

Daphne-HU, 2025-01-15:
I have attempted the suggested solutions, including using different power supplies and PCBs, but the issue persists. I also tried programming a different BLOCK_KEY without success. Could there be other potential causes or solutions to this problem?

Daphne-HU, 2025-01-21:
@KonstantinKondrashov
Thanks for your response!
I’m seeing different cases on different devices, some work some don't.
On most devices that do not work i get case 1. Nothing is burned in BLOCK_KEY4.
All the devices have the same PCB design.

Do you have any idea what hardware issue this could be?
I tried powering it with a stable external power supply, but that didn’t make any difference.

KonstantinKondrashov, 2025-02-05:
Hi @Daphne-HU!

Sorry, but I do not have any idea. It looks like it is not a software issue. 

> Possible hardware-related issues:
>
> ESD (Electrostatic Discharge)
> Weak Power Supply

dobairoland, 2025-02-10:
I'm closing this because I don't think we can do anything about this on the esptool side and don't see a reason to track an individual issue.

closed 2025-02-10 by dobairoland