espressif/esptool #1050: Esptool 4.8.1 cannot get official Devkitc (WROVER-E) in bootmode

sim0njo, 2025-01-11:
### Operating System

Windows10

### Esptool Version

4.81

### Python Version

3.11.2

### Chip Description

ESP32 WROVER-E

### Device Description

Official ESP-DEVKITC-VE on custom PCB

### Hardware Configuration

GPIO 32/33/21/18/19/27/26/25/34

### How is Esptool Run

From ESP-IDF plugin for Visual Studio Code OR from ESP-IDF Terminal

### Full Esptool Command Line that Was Run

C:\....\esp\v5.3.2\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM3', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '40m', '--flash_size', 'detect', '0x1000', 'bootloader/bootloader.bin', '0x10000', 'e32P2M-4Mb-pV3.bin', '0x8000', 'partition_table/partition-table.bin', '0xd000', 'ota_data_initial.bin'

### Esptool Output

```shell
esptool.py v4.8.1
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### More Information

I use the official Espressif ESP32-DEVKITC-VE, when working with the old ESP-IDF 4.4 under Eclipse (esptool.py v3.0-dev) this module is flashable in 99,9% of the cases when seated onto my custom PCB. So no issues there.

Now I am upgrading to ESP-IDF 5.3.2 under Visual Studio Code and esptool.py v4.8.1 is not able to switch the ESP32 to boot mode., although it is possible to read the ESP32. Esptool is just not handling the RTS/CTS signals correctly to force boot mode.

So now I force the DEVKITC into boot mode by hand when esptool is in the connecting phase, now it is able to flash the ESP32.

===
Connecting................................
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:b6:1f:ed:54:fc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Flash will be erased from 0x00010000 to 0x00101fff...
Flash params set to 0x0230
SHA digest in image updated
Compressed 25328 bytes to 15641...
Wrote 25328 bytes (15641 compressed) at 0x00001000 in 0.7 seconds (effective 281.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.1 seconds (effective 270.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective 494.4 kbit/s)...
Hash of data verified.
Compressed 990096 bytes to 645710...
Wrote 990096 bytes (645710 compressed) at 0x00010000 in 16.3 seconds (effective 487.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

Honza0297, 2025-01-14:
Hello @sim0njo! Do you know how to use `git bisect` (https://git-scm.com/docs/git-bisect)? It can help to find the exact commit when the flash stopped working.

sim0njo, 2025-01-14:
Git bisect on which sources?

sim0njo, 2025-01-14:
I had a look at the different commits between v3.0 and 4.8.1... doesn't say anything to me...

Honza0297, 2025-01-14:
Thank you for your cooperation so far! The idea is to `git bisect` esptool repository and find in which commit the flash stops working.  The reason I am not doing it myself is that, unfortunately, I cannot replicate your issue (with just the devkit). However, If we manage to find the commit where the flash stops working, I will be able to identify the culprit and fix it even without replicating the issue. 

If are are still willing to help, I can write a step by step guide what to do

sim0njo, 2025-01-14:
Ok, tell me

Honza0297, 2025-01-15:
Thanks, here's the guide:
1) Build your application.
2) Open new terminal window, clone esptool repo and install it as an editable package (`pip install -e <path-to-esptool-repo>`). 
2)  Being in the esptool repository folder,  start git bisect with `git bisect start`.
3) Choose a bad commit. We know it fails with esptool 4.8.1, so this will be our bad commit: `git bisect bad 1e9df29` (release of 4.8.1 version).
4) Choose a good commit. That will be esptool 3.0: `git bisect good 4fa0bd7` (release of 3.0 version). 
5) Now, in a terminal window used in step 1, navigate to your app's build folder (`<path to your app>/build`) and try to flash your ESP32 with this command: `esptool.py -p <port> -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader/bootloader.bin 0x10000 e32P2M-4Mb-pV3.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin`. Note that you need to specify correct port. 
6) If the flash will be successful, write `git bisect good` into the esptool terminal window. By this, you notify git bisect that the current commit is good. git bisect will automatically switch to the next commit to test.  If the flash will be unsuccessful, write `git bisect bad`.  
7) Repeat steps 5 and 6 until you find the commit when the flash stops working. It will take approx. 8 steps. 
8) Git bisect will finally show you the first bad commit. Please, copy its commit hash.

sim0njo, 2025-01-15:
Ok following your guide above I find below commit to be the problem:

simonjo@C-xxxxxxx MINGW64 ~/sources/esptool ((a692fe4...)|BISECTING)
$ git bisect good a692fe4
3e6cf6403ac454c3e1f84f2c938feca68f5fc815 is the first bad commit
commit 3e6cf6403ac454c3e1f84f2c938feca68f5fc815
Author: radim.karnis <radim.karnis@espressif.com>
Date:   Mon Jun 14 15:34:15 2021 +0200

    esp32s3: Add USB-JTAG-Serial support

 esptool.py                           | 147 ++++++++++++++++++-----------------
 flasher_stub/Makefile                |   2 +-
 flasher_stub/include/rom_functions.h |   8 ++
 flasher_stub/include/soc_support.h   |  20 ++++-
 flasher_stub/stub_io.c               |   8 +-
 5 files changed, 109 insertions(+), 76 deletions(-)

Honza0297, 2025-01-17:
Thanks for clarification! I supposed you built the application in terminal 1 and thus had ESP IDF environment set up in git bash number 1. But if esptool changed its version in each iteration, it means everything was OK and the correct esptool was tested. 

In the meantime, I did give it a shot and analyzed the commit you posted. There are no esp32 related changes in that specific commit, **but** sometime in that time period, a timing workaround for esp32r0 was removed and maybe your specific setup was coincidentally working thanks to this (yet completely unrelated!) workaround. 

To check my theory, can you please put a 10 uF capacitor between EN and GND pins and try to program your ESP32 with some newer version of esptool (preferably the latest)?

sim0njo, 2025-01-17:
Ok, I removed the EN pushbutton and soldered a 10uF/50V capacitor on the Devkitc board, then tried to flash it -> works !

Executing task in folder e32Phc2Mqtt: C:\Users\simonjo\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe C:\Users\simonjo\esp\v5.3.2\esp-idf\components\esptool_py\esptool\esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader/bootloader.bin 0x10000 e32P2M-4Mb-pV3.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 

esptool.py v4.8.1
Serial port COM3
Connecting.........
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:b6:1f:ed:54:fc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x00101fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Flash params set to 0x0230
SHA digest in image updated
Compressed 25328 bytes to 15637...
Wrote 25328 bytes (15637 compressed) at 0x00001000 in 0.7 seconds (effective 284.7 kbit/s)...
Hash of data verified.
Compressed 987504 bytes to 644177...
Wrote 987504 bytes (644177 compressed) at 0x00010000 in 16.0 seconds (effective 493.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.1 seconds (effective 293.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective 555.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Honza0297, 2025-01-20:
Glad to hear it works now! 

If you run into these issues in the future, you can also try to write custom reset sequence as described  in the [documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html#custom-reset-sequences).  

For inspiration: the sequence below corresponds to the "workaround" sequence I mentioned in the previous comment. When compared to the standard one, it has longer wait times. 

```
[esptool]
custom_reset_sequence = D0|R1|W1.3|D1|R0|W0.5|D0
```

This issue will be closed in a short time, but feel free to reopen/open new issue if you run into any problems!

closed 2025-01-20 by dobairoland

sim0njo, 2025-01-23:
I will try it when I get back home

sim0njo, 2025-01-24:
I tried it, and sometimes it works (2 out of 10)... so not really workable

Honza0297, 2025-01-26:
Have you tested the sequence on the ESP with the capacitor added? 
In that case, maybe the wait times are too long.

sim0njo, 2025-01-26:
I first restored the Devkitc v4 in it's original state, so removing the 10uF cap and reinstalling the pushbutton.

It is purely depending on the timing of the DTR/RTS signals, the components on the board are ok as this was working 99.9% of the time before. 

Can you point me to the commit you earlier mentioned
>>>
In the meantime, I did give it a shot and analyzed the commit you posted. There are no esp32 related changes in that specific commit, but sometime in that time period, a timing workaround for esp32r0 was removed and maybe your specific setup was coincidentally working thanks to this (yet completely unrelated!) workaround.
<<<

Honza0297, 2025-01-27:
I [can](https://github.com/espressif/esptool/commit/2d72f56ce6b6103d908194f6d4237a2032ce9b89). 
**TLDR**: Early versions of ESP32 chips needed longer waiting times between DTR and RTS changes.  Every other time, esptool tried to connect to the chip using this workaround when it prolonged waiting times, no matter which chip revision was connected. 

This led me to the idea with the capacitor (which worked) and thus i supposed a custom reset sequence should work as well. I still think it, but you probably need to fine tune the waiting times. Unfortunately, I cannot provide exact waiting times which will work for 100 %, because they are dependent on your exact setup (combination of OS, driver, chip and their versions/revisions).