espressif/esptool #1051: ESP-32S can not flash and reset

Voffkait, 2025-01-14:
### Operating System

Windows 10

### Esptool Version

4.8.1

### Python Version

Python 3.12.4

### Chip Description

ESP-32S

### How is Esptool Run

Windows terminal

### Esptool Output

```shell
C:\Users\HUNTE>python -m esptool --chip esp32 erase_flash
esptool.py v4.8.1
Found 1 serial ports
Serial port COM12
COM12 failed to connect: Could not open COM12, the port is busy or doesn't exist.
(could not open port 'COM12': OSError(22, 'Указано несуществующее устройство.', None, 433))

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.
```

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

radimkarnis, 2025-01-14:
Hi @Voffkait,

this is not an esptool issue, but OS/driver issue. Windows fails to open the `COM12` port, without it esptool cannot communicate with the chip.

Please try to re-install the correct drivers for the USB-UART bridge on your devkit (most likely cp2102).

You can verify the connection by using any [serial terminal program](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#serial-terminal-programs), like PuTTY on Windows. If you don't manage to connect to the port and see the data coming out of the chip, esptool cannot work.

Voffkait, 2025-01-14:
esptool.py v4.6
Serial port COM11
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html

Now I have that error then trying to flash it from Arduino IDE

radimkarnis, 2025-01-14:
@Voffkait have you read the [Troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html)? Many reasons can cause this error, most of which are mentioned in the guide, along with possible solutions.

No data is coming from your ESP32. That might mean it is not switching to download mode automatically, or it's being held in reset. Do you see any data when using PuTTY and pressing the RESET button?

I am sorry, but the GitHub issue tracker should be used to report bugs in the tool itself. We can't be troubleshooting cases like this, especially when no additional info is provided.

radimkarnis, 2025-01-14:
I see you've deleted your comments. I believe that means your issues have been resolved. I will close this now. Please reopen only if you are certain that this is a bug in esptool and you've tried all the steps in the troubleshooting guide.

closed 2025-01-14 by radimkarnis