espressif/esptool #1052: 高版本ESPTOOLS烧录提示MD5校验错误

Exculivor, 2025-01-15:
### Operating System

Win11

### Esptool Version

v4.7.dev3

### Python Version

py3.11

### Chip Description

ESP32S3R8V

### Device Description

ESP32S3R8V+GD25LQ256DWIGR

### Hardware Configuration

[SCH.pdf](https://github.com/user-attachments/files/18418154/SCH.pdf)

### How is Esptool Run

VSCODE

### Full Esptool Command Line that Was Run

 C:\Users\Administrator\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\Administrator\.espressif\frameworks\esp-idf-v5.1.2\components\esptool_py\esptool\esptool.py -p COM51 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dout --flash_freq 80m --flash_size detect 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin

### Esptool Output

```shell
esptool.py v4.7.dev3
Serial port COM51
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_1v8)
Crystal is 40MHz
MAC: 7c:df:a1:e4:68:c0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 32MB
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00041fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 22384 bytes to 14203...
Wrote 22384 bytes (14203 compressed) at 0x00000000 in 0.5 seconds (effective 363.7 kbit/s)...
File  md5: ea91be95071854f4b33eed1547b0a0a3
Flash md5: d0913d86e665dac358d3829b98c691e6
MD5 of 0xFF is d0913d86e665dac358d3829b98c691e6

A fatal error occurred: MD5 of file does not match data in flash!

 *  终端进程“C:\Users\Administrator\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe 'C:\Users\Administrator\.espressif\frameworks\esp-idf-v5.1.2\components\esptool_py\esptool\esptool.py', '-p', 'COM51', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s3', 'write_flash', '--flash_mode', 'dout', '--flash_freq', '80m', '--flash_size', 'detect', '0x0', 'bootloader/bootloader.bin', '0x10000', 'hello_world.bin', '0x8000', 'partition_table/partition-table.bin'”已终止，退出代码: 2。
```

### More Information

硬件：ESP32S3R8V+GD25LQ256DWIGR
工具：flash_download_tool_3.9.7、flash_download_tool_3.9.4
IDE：VS CODE + ESP-IDF V5.1.2、esptool.py v4.7.dev3
现象：使用flash_download_tool_3.9.4可以正常烧录，使用VSCODE或者使用flash_download_tool_3.9.7下载都提示MD5校验失败。把flash_download_tool_3.9.7的spi_download.conf里面的verify = True改成False能正常下载。
[ESP32下载失败图片.zip](https://github.com/user-attachments/files/18418166/ESP32.zip)

### I Have Read the Troubleshooting Guide

- [X] I confirm I have read the troubleshooting guide.

dobairoland, 2025-01-15:
Hi @Exculivor. Could you please try to use esptool directly instead of the flash_download_tool?

Here is a guide how to set up a separate esptool installation: https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html#virtual-environment-installation. Please install the latest version of esptool, version 4.8.1. Are you experiencing the same issue with it? If yes then please share the logs after adding the `--trace` command line argument.

Exculivor, 2025-01-17:
[log.txt](https://github.com/user-attachments/files/18447932/log.txt)

Hi, I have updated the version as you suggested, but the problem still persists. Here is the log I captured as you instructed. Please help me check where the problem lies. Thank you for your efforts!

Exculivor, 2025-01-17:
[log_esptool.py v4.6.2.txt](https://github.com/user-attachments/files/18447975/log_esptool.py.v4.6.2.txt)

This is the log after I downgraded ESPTOOLS to version 4.6.2.

peterdragun, 2025-01-21:
Hi @Exculivor,
I am sorry but we are not able to reproduce your issue using 32MB flash in quad mode (but unfortunately we don't have the exact combination of chips available). The version that didn't fully support flash sizes bigger than 16MB seems to work for you, so I would guess that something that was added as part of that support has caused the issue for you. Maybe you can try running the same command with `--no-stub` on the newest esptool?
There is one thing that is common for both of your runs - even in the successfull run there seems to be an issue with reading the checksum. You can see it in the monitor output that the SHA256 does not match with the expected value. Similar thing happend in the newest esptool run where the MD5 checksum was not matching and was result of failed run. I am wondering if the app was actually flashed without any issue but flashing failed because the checksum did not match for some reason.

Are you familiar with the `git bisect` command, and are you willing to help us determine the root cause of this issue?

dobairoland, 2025-02-03:
I'm closing this for inactivity. We can reopen it if we receive more information based on the previous comment.