espressif/esptool #1056: Batch burn program on Linux

gengge, 2025-01-22:
### Is your feature request related to a problem?

I can batch burn programs in factory mode on Windows. But I want to bulk burn programs for the esp32 on Linux. Are there any relevant commands and how do I operate?

radimkarnis, 2025-01-24:
Hello @gengge,

I don't know about any program for Linux using esptool to bulk program a batch of ESPs.

It is quite easy to [use esptool in a Python script](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/scripting.html). A simple script going through all ports and running the 'write_flash' operation on each of them one-by-one could be the solution. Multiple operations could even run at the same time if multithreading was used to speed things up.

radimkarnis, 2025-02-14:
Closing this for inactivity. Please feel free to reopen if you need more information.

closed 2025-02-14 by radimkarnis