espressif/esptool #1057: image_info output can cause issues with Bash

KaeLL, 2025-01-23:
### Operating System

Linux

### Esptool Version

4.9.dev4

### Python Version

Python 3.13.1

### Full Esptool Command Line that Was Run

VAR=$(esptool.py image_info -v 2 build/app.bin | tr '\0' '?')

### Esptool Output

```shell
...
Application information
=======================
Project name: app??????????
App version: 1.2.3???????????????????????????
Compile time: Jan 17 2025????? 14:43:46????????
...
```

### What is the Expected Behaviour?

The output from the binary header should be sanitized because Bash doesn't handle NUL bytes particularly well in strings, variables, and it can cause problems for scripts.
In this case, it Bash complains: `bash: warning: command substitution: ignored null byte in input`

radimkarnis, 2025-01-24:
Hello @KaeLL,
thanks for the report!

We'll sanitize the app info and bootloader desc.

radimkarnis, 2025-09-22:
The latest IDF still uses esptool `v4` (it's constrained as `esptool~=4.10`). Adopting `v5` is still in the process, as we need to accommodate all of the breaking changes and keep offering support for `v4` during the changeover period.