espressif/esptool #1065: Addition to Troubleshooting docs

eimermusic, 2025-02-14:
### Is your feature request related to a problem?

I ran into a serial exception when uploading to an ESP32 device I am developing for. The exat same error as at the bottom of the troubleshooting guide at: https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#a-serial-exception-error-occurred

`
A serial exception error occurred: read failed: [Errno 6] Device not configured
`

My root cause was none of the listed causes but still fairly basic one... 

### Describe the solution you'd like

I had bloated my build so that it exceeded the device capacity. By removing a graphic asset (as a header file) I had just added... my device was back on and happy. Might be worth adding a note about this cause to that section.

Honza0297, 2025-02-17:
Hello @eimermusic and thank you for your report! 
Can you please provide more information/circumstances about your application? Both `ESP-IDF` and `esptool.py` shoudl have checks to prevent loading binaries bigger than memory size. 
Thank you!

Honza0297, 2025-03-03:
Without additional information, we unfortunately cannot add any specific case causing the aforementioned error to the troubleshooting guide. 
Closing this for inactivity.

closed 2025-03-03 by Honza0297