espressif/esptool #1066: Customer Logger Redirect Issue

JeremiahGillis, 2025-02-19:
Thanks for adding the custom logger. I tested this today and found an issue using your example function:

https://github.com/espressif/esptool/blob/6d04155e886ae65bdcb28a7993b43f2476f6ce16/docs/en/esptool/scripting.rst?plain=1#L102-L104

This will result in an error:

```
[CustomLogger]: esptool.py v4.8.1
[CustomLogger]: Serial port COM4
[CustomLogger]: Connecting...[CustomLogger]: 
[CustomLogger]: Detecting chip type...[CustomLogger]:  ESP32-S3
[CustomLogger]: Chip is ESP32-S3 (QFN56) (revision v0.1)
[CustomLogger]: Features: WiFi, BLE, Embedded PSRAM 8MB (AP_1v8)
[CustomLogger]: Crystal is 40MHz
[CustomLogger]: USB mode: USB-Serial/JTAG
[CustomLogger]: MAC: 7c:df:a1:e3:f1:d0
[CustomLogger]: Enabling default SPI flash mode...
[CustomLogger]: Configuring flash size...
[CustomLogger]: Auto-detected Flash size: 32MB
[CustomLogger]: Flash will be erased from 0x00000000 to 0x00004fff...
[CustomLogger]: Flash will be erased from 0x00008000 to 0x00008fff...
[CustomLogger]: Flash will be erased from 0x00010000 to 0x003a4fff...
[CustomLogger]: Erasing flash...
[CustomLogger]: SHA digest in image updated
[CustomLogger]: Compressed 18880 bytes to 12178...
[CustomLogger]: Took 0.15s to erase flash block
[CustomLogger]: Writing at 0x00000000... (8 %)
[CustomLogger]: Writing at 0x000007b1... (16 %)
[CustomLogger]: Writing at 0x00001254... (25 %)
[CustomLogger]: Writing at 0x00001860... (33 %)
[CustomLogger]: Writing at 0x00001e42... (41 %)
[CustomLogger]: Writing at 0x0000240a... (50 %)
[CustomLogger]: Writing at 0x000029a6... (58 %)
[CustomLogger]: Writing at 0x00002f24... (66 %)
[CustomLogger]: Writing at 0x00003422... (75 %)
[CustomLogger]: Writing at 0x00003923... (83 %)
[CustomLogger]: Writing at 0x00003dce... (91 %)
[CustomLogger]: Writing at 0x000043f5... (100 %)
Error calling Python override of QThread::run(): Traceback (most recent call last):
  File "C:\Users\jerem\Source\Repos\removed\programming.py", line 159, in run
    self.flash_mcu()
  File "C:\Users\jerem\Source\Repos\removed\programming.py", line 120, in flash_mcu
    esptool.main(cmd)
  File "c:\Users\jerem\Source\Repos\removed\.venv\Lib\site-packages\esptool\__init__.py", line 1047, in main
    operation_func(esp, args)
  File "c:\Users\jerem\Source\Repos\removed\.venv\Lib\site-packages\esptool\cmds.py", line 707, in write_flash
    log.print_overwrite(
TypeError: CustomLogger.print_overwrite() got an unexpected keyword argument 'last_line'
```
The example function should be:

```python
def print_overwrite(self, message, last_line=False):
    # Overwriting not needed, print normally
    self.print(message)
```
This fixes my error and allows the device to be fully programmed.

JeremiahGillis, 2025-02-21:
@radimkarnis The new commit causes an issue elsewhere:

```
Error calling Python override of QThread::run(): Traceback (most recent call last):
  File "C:\Users\jerem\Source\Repos\Zenith\zti-test-pylink-gui\programming.py", line 77, in run
    self.flash_mcu()
  File "C:\Users\jerem\Source\Repos\removed\programming.py", line 69, in flash_mcu
    esptool.main(cmd)
  File "c:\Users\jerem\Source\Repos\removed\.venv\Lib\site-packages\esptool\__init__.py", line 1047, in main
    operation_func(esp, args)
  File "c:\Users\jerem\Source\Repos\removed\.venv\Lib\site-packages\esptool\cmds.py", line 637, in write_flash
    log.print_overwrite(
TypeError: CustomLogger.print_overwrite() missing 1 required positional argument: 'last_line'
```

Some print_overwrite functions pass `last_line` while others do not, for example:

https://github.com/espressif/esptool/blob/eaaa6b37fbdca6d1a9481f73f7b1f055bc44d39c/esptool/cmds.py#L637-L639

https://github.com/espressif/esptool/blob/eaaa6b37fbdca6d1a9481f73f7b1f055bc44d39c/esptool/cmds.py#L707-L711

This is why my suggest fix has to have `last_line=False` for cases where `last_line` is not passed.

radimkarnis, 2025-02-21:
🤦 sorry for that, you are right. ok, let's do another round

closed 2025-02-25 by espressif-bot