espressif/esptool #1070: User code causes subsequent flashing to fail unless bootloader is entered manually

thegoodhen, 2025-03-09:
### Operating System

Windows 10

### Esptool Version

esptool.py v4.6 4.6

### Python Version

3.10.11

### Chip Description

ESP32S3

### Device Description

Mutliple different boards (custom, official devkits...) when using USB CDC output in user program. 

[image]

### Hardware Configuration

Not using UART, using hardware CDC only

### How is Esptool Run

Both from Arduino IDE and directly from commandline

### Full Esptool Command Line that Was Run

D:\work\stavebnice\debugusb>"C:\Users\thego\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6/esptool.exe" --chip esp32s3 --port "COM24"  --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x0 "C:\Users\thego\AppData\Local\Temp\arduino\sketches\381CE3FFAA9043143049D630B96A721B/serialBugTest6.ino.bootloader.bin" 0x8000 "C:\Users\thego\AppData\Local\Temp\arduino\sketches\381CE3FFAA9043143049D630B96A721B/serialBugTest6.ino.partitions.bin" 0xe000 "C:\Users\thego\Documents\Arduino\hardware\gallino\esp32/tools/partitions/boot_app0.bin" 0x10000 "C:\Users\thego\AppData\Local\Temp\arduino\sketches\381CE3FFAA9043143049D630B96A721B/serialBugTest6.ino.bin"

### Esptool Output

```shell
esptool.py v4.6
Serial port COM24
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010804000707122000000000c0
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.003 Read 98 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0c0010804 | ...... .........
    0007071220000000 00c0c00108040007 | .... ...........
    07122000000000c0 c001080400070712 | .. .............
    2000000000c0c001 0804000707122000 |  ............. .
    000000c0c0010804 0007071220000000 | ............ ...
    00c0                              | ..
TRACE +0.000 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000
TRACE +0.002 Received full packet: 010804000707122000000000

TRACE +0.002 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=00100040
TRACE +0.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE +0.001 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000900000000000000c0
[... 844 lines cut ...]

```

void setup() {

  Serial.begin(115200);
  //delay(1000);//uncommenting this like fixes the problem
}

void loop() {
  Serial.write(0x0D);
  Serial.write(0x0E);
  Serial.write(0x0A);
  Serial.write(0x0D);
  Serial.write(0x0B);
  Serial.write(0x0E);
  Serial.write(0x0E);
  Serial.write(0x0F);
}

```

### I Have Read the Troubleshooting Guide

- [x] I confirm I have read the troubleshooting guide.

radimkarnis, 2025-03-16:
Hi @thegoodhen,

thanks for the detailed report and steps to reproduce. As mentioned in the linked ticket and https://github.com/espressif/esptool/issues/868, we don't know what causes this, but all points to a possible fix in the flasher stub. We will try to give this a proper look soon!

thegoodhen, 2025-03-18:
Hello @radimkarnis No problem! So far I have managed to use a workaround (--no-stub) so I have been happy, but I also do believe the problem to be in a flasher stub. If you need me to look into it further, let me know. It was one of two critical problems in my project, the other being the fact that the ESP32 compilation on Arduino is taking way too long on regular computers (welp, that will take a bit of effort to figure out...). But relying on no-stub might slow down the upload process in theory (? haven't checked...) so I think it would still be good to get it resolved. Let me know if you need to test anything or look into the stub implementation for potential issues!

tjhayko, 2025-05-13:
Has this issue been addressed?  

I'm having what seems like a similar problem, although it only fails on one machine (my main development computer) but works fine on a laptop.

I've tried adding --no-stub workaround, but that doesn't seem to work for me either.

tjhayko, 2025-09-26:
I'm actually not sure of the answers to your questions.

I think I'm using USB-CDC, as I'm definitely not using a JTAG board

The board in question is a Sequent Microsystems ESP32-Pi [(https://sequentmicrosystems.com/products/esp32-pi-low-cost-replacement-for-raspberry-pi?_pos=1&_psq=ESP&_ss=e&_v=1.0)](https://sequentmicrosystems.com/products/esp32-pi-low-cost-replacement-for-raspberry-pi?_pos=1&_psq=ESP&_ss=e&_v=1.0) , which I think from board schematic has GPIO0 connected to DTR, and EN connected to RTS, both on CH340.

On the computer where it works, I don't do anything other than initiate the download using esptool.  I normally use Visual Studio Code with Platform.io for this, but I've also tried it from the Arduino IDE and by manually running esptool, and the behavior consistent.  The slower one (a laptop) it mostly works, but sometimes fails.  On the faster one (a desktop), it has never worked.  I sort of knew that GPIO0 and EN need to be strapped for the download to work, and since it usually works on my laptop, I figured the strapping was handled mostly correctly.

I'll give a try with one of the DOIT DevKit clones I've got as well to see if the problem occurs.

Sorry, I'm still very much learning this stuff, but I've got a multimeter, and I'm very willing to measure/debug whatever I can that might help.

tjhayko, 2025-09-26:
I will do.  I will likely not get to it for a day or two though.  Should I open a different issue once I have those logs captured?

jedi7, 2025-09-26:
@tjhayko, to avoid confusion for the team, let's create a new ticket for this. The new issue should fully capture the investigation from this thread. Could you please create it and summarize all the key findings discussed here? Thanks!