espressif/esptool #1072: esptool complement not loading correctly

Tinyu-Zhao, 2025-03-11:
### Operating System

macOS 14.7.4

### Esptool Version

4.8.1

### Python Version

3.13.2

### Full Esptool Command Line that Was Run

esptool.py xxxxxxx

### Esptool Output

```shell

```

### More Information

The esptool I installed via brew install esptool prompts me when I press tab after entering any parameter or part of a parameter.

However, after importing the idf environment by executing the following command, the autocomplete feature of esptool disappears when I press tab after typing any parameter or part of a parameter.
`$HOME/Developer/Code/uiflow_micropython/esp-idf/export.sh`.
or
`. $HOME/Developer/esp/master/esp-idf/export.sh`.

But any other command will work fine, for example, just type ls /dev/cu. and press tab and the serial port will show up.

I tried uninstalling the local global esptool by brew uninstall esptool and restarting the terminal, then importing the idf environment, the esptool is patched, why is that, the esptool versions are all the same!

peterdragun, 2025-03-17:
Hi @Tinyu-Zhao, please note that we are not maintaining the package from `brew`, if you have any issues with that I would recommend opening the issue in the brew repository: https://github.com/Homebrew/homebrew-core.

Even if the package installed by `brew` and one from ESP-IDF are the same version they may not work correctly as they are installed using two different package managers in two different locations. I would suggest using the esptool within your esp-idf environment to avoid this clashing or if you want to install it globally use `pip`, as suggested by our documentation: https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html#global-installation

Tinyu-Zhao, 2025-03-18:
> Hi [@Tinyu-Zhao](https://github.com/Tinyu-Zhao), please note that we are not maintaining the package from `brew`, if you have any issues with that I would recommend opening the issue in the brew [...]

Thanks for your help.

closed 2025-03-18 by Tinyu-Zhao