espressif/esptool #1081: Can’t Flash esp12F

wohdek21, 2025-04-21:
### Operating System

Windows 11

### Esptool Version

4.8.1

### Python Version

Python 3.13.2

### Chip Description

ESP8266

### Device Description

Plain module, no breadboard

### Hardware Configuration

Nothing, setup for flashing
GPIO 0, 15 pulled down
GPIO 2, RST, EN pulled up

Powering through external 3.3v power source

### How is Esptool Run

Terminal, Arduino IDE

### Full Esptool Command Line that Was Run

"C:\Users\xonix\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3" -I "C:\Users\xonix\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/upload.py" --chip esp8266 --port "COM4" --baud "115200" ""  --before default_reset --after hard_reset write_flash 0x0 "C:\Users\xonix\AppData\Local\arduino\sketches\4FDF8A02E37BBBCA2EA860DB6615B13F/Blink.ino.bin"

### Esptool Output

```shell
. Variables and constants in RAM (global, static), used 28104 / 80192 bytes (35%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1496     initialized variables
╠══ RODATA   920      constants       
╚══ BSS      25688    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 59667 / 65536 bytes (91%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     26899    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 232148 / 1048576 bytes (22%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     232148   code in flash   
"C:\Users\xonix\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3" -I "C:\Users\xonix\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/upload.py" --chip esp8266 --port "COM4" --baud "115200" ""  --before default_reset --after hard_reset write_flash 0x0 "C:\Users\xonix\AppData\Local\arduino\sketches\4FDF8A02E37BBBCA2EA860DB6615B13F/Blink.ino.bin"
esptool.py v3.0
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal esptool.py error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
```

### More Information

ESP successfully boots into mode 1, 0

### Other Steps to Reproduce

Nothing

### I Have Read the Troubleshooting Guide

- [x] I confirm I have read the troubleshooting guide.

wohdek21, 2025-04-22:
None of this working for me, but UART and console is working

Jason2866, 2025-04-23:
Not using esptool.py v4.8.1. Your log and command line shows something different.
```
"C:\Users\xonix\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3" -I "C:\Users\xonix\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/upload.py" --chip esp8266 --port "COM4" --baud "115200" ""  --before default_reset --after hard_reset write_flash 0x0 "C:\Users\xonix\AppData\Local\arduino\sketches\4FDF8A02E37BBBCA2EA860DB6615B13F/Blink.ino.bin"
esptool.py v3.0
```
Update your ArduinoIDE and all tools. If this does not help deinstall ArduinoIDE and install fresh.

Honza0297, 2025-04-25:
Hello @wohdek21!
If your ESP12F enters boot mode (1,0) (which is flash boot mode), it most likely means that there is some issue with strapping pins. This is most likely not an esptool issue, but rather a HW one.
I see you listed all the strapping pins and their correct values. If you are 100 % sure they are also correctly soldered/connected physically, can you please post a photo of your circuit, including the USB-serial adapter?

wohdek21, 2025-04-28:
Seems like error was with autorst, bought wemos board and going to use it for uploading OTA firmware
Thanks to everybody for help!

closed 2025-04-28 by wohdek21