espressif/esptool #1083: idf-5.5 ESP32 stub loader #1 crash on D2(READ_FLASH) command (ESPTOOL-1056)

GooogLeMorons, 2025-04-30:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Tx: [00 D2 10 00 00 00 00 00 00 10 00 00 00 02 00 00 20 00 00 00 00 00 00 00]
Rx: [01 D2 02 00 00 00 00 00 00 00]

### Actual behavior (suspected bug)

Rx:
Fatal exception (0): IllegalInstruction
epc1=0x40006843, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. Boot ROM loader
2. Upload stub loader from ${UserHome}.espressif\python_env\idf5.5_py3.11_env\Lib\site-packages\esptool\targets\stub_flasher\1\ESP32.json
3. Launch it, ensure Rx SLIP-wrapped magic value [4F 48 41 49]
4. Issue (SLIP-wrapped) 0xD2 command, e.g. [C0 00 D2 10 00 00 00 00 00 00 10 00 00 00 02 00 00 20 00 00 00 00 00 00 00 C0]
5. Enjoy 'Fatal exception' raw response instead of SLIP-wrapped command result

### Project release version

5.5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

doesn't matter

### Shell

other (details in Additional context)

### Additional context

it suxx extremely, since ability "to read" is a must to ensure "writing" is fine

radimkarnis, 2025-05-05:
Hi @GooogLeMorons,

Without any additional information on how you load the stub flasher and how you send the `(SLIP-wrapped) 0xD2 command`, I cannot give you any support, except for the following pointers:

- [Serial protocol description](https://docs.espressif.com/projects/esptool/en/latest/advanced-topics/serial-protocol.html) -  please check the packets you send against the descriptions in this document.
- In your other related issue ticket https://github.com/espressif/esptool/issues/1084, you mention you use the stub files from the following directory `\stub_flasher\2\ESP32.json`. As mentioned in the previously opened ticket https://github.com/espressif/esp-flasher-stub/issues/12, these are deprecated and only for testing purposes and won't receive any updates or support in the future.
- Run any esptool flash operation (e.g., `read-flash`) with the `--trace` argument - you will see the commands do work. The trace also contains all information about the packets being sent. With this info, you can work to reproduce the behavior.

radimkarnis, 2025-05-06:
Please try to study the trace output of esptool and the serial protocol to proceed with your use case. I will close the tickets in the meantime.

closed 2025-05-06 by radimkarnis