espressif/esptool #1085: Support for external SPI memories bigger than 16MB

RELOC-DEV, 2025-05-02:
### Is your feature request related to a problem?

We are trying to use the esptool to preload some data into an external SPI flash (manually configuring the SPI pins with the "--spi-connection" parameter).
The operation works fine with images < 16 MB but always fails with bigger images checking the MD5.

### Describe the solution you'd like

Debugging a bit with an oscilloscope I noticed that the ESP stub always uses the SPI commands for 24bit address instead of the 32bit version, de-facto limiting to 16777215 byte (16MB) the addressable size.
However I suspect that this is not the only "missing-element" for the feature.

Are there any plans to support external (but even internal I suppose) flash bigger than 16 MB?

### Describe alternatives you've considered

We have two options:
1) modify the esptool to extend the support (very hard option)

2) Create our custom solution (the only actual option)

radimkarnis, 2025-05-02:
Hello @RELOC-DEV,
we do support 32-bit addressing (flash sizes >16MB) on the ESP32-S3 and now also on the ESP32-P4 when the flasher stub is used. What chip type do you use?

The support should work for both internal and external flash chips.

Please look at the recently released stub flasher with >16MB support for the P4: https://github.com/espressif/esptool-legacy-flasher-stub/releases/tag/v1.5.0

RELOC-DEV, 2025-05-05:
Thank you @radimkarnis for your useful reply.
Luckily, we are using the ESP32-S3.

However, I tried the v1.5.0 of the esptool-legacy-flasher-stub, making sure that it actually use it (deleting the esp32s3.json results in the error "Stub flasher JSON file for esp32s3 not found.") but it always uses the 24bit commands.

I used the following commands:

Erase all the flash:
```
python.exe esptool.py -c esp32s3 -p COM7 -b 921600 erase-flash --spi-connection 45,48,47,3,14

esptool.py v4.8.1
Loaded custom configuration from D:\2024-SICOR\ext_flash_load\esptool-git\esptool\esptool.cfg
Serial port COM7:
Connecting...
Connected to ESP32-S3 on COM7:
Chip type:          ESP32-S3 (QFN56) (revision v0.2)
Features:           Wi-Fi, BT 5 (LE), Dual Core + LP Core, 240MHz, Embedded PSRAM 8MB (AP_3v3)
Crystal frequency:  40MHz
USB mode:           USB-Serial/JTAG
MAC:                dc:da:0c:77:19:98

Uploading stub flasher...
Running stub flasher...
Stub flasher running.
Changing baud rate to 921600...
Changed.

Configuring SPI flash mode (CLK:45, Q:48, D:47, HD:3, CS:14)...
Erasing flash memory (this may take a while)...
Flash memory erased successfully in 127.9 seconds.
```

Write some data to address 0x0100 0000 so that 32bit address is required:

```
python.exe esptool.py -c esp32s3 -p COM7 -b 921600 write_flash --spi-connection 45,48,47,3,14 --flash-size detect 0x1000000 data.exe

Warning: Deprecated: Command 'write_flash' is deprecated. Use 'write-flash' instead.
esptool.py v4.8.1
Loaded custom configuration from D:\2024-SICOR\ext_flash_load\esptool-git\esptool\esptool.cfg
Serial port COM7:
Connecting...
Connected to ESP32-S3 on COM7:
Chip type:          ESP32-S3 (QFN56) (revision v0.2)
Features:           Wi-Fi, BT 5 (LE), Dual Core + LP Core, 240MHz, Embedded PSRAM 8MB (AP_3v3)
Crystal frequency:  40MHz
USB mode:           USB-Serial/JTAG
MAC:                dc:da:0c:77:19:98

Uploading stub flasher...
Running stub flasher...
Stub flasher running.
Changing baud rate to 921600...
Writing at 0x01000000 [                              ]   0.0% 0/598245 bytes...
Writing at 0x010066fa [                              ]   2.7% 16384/598245 bytes...
Writing at 0x0100d283 [>                             ]   5.5% 32768/598245 bytes...
Writing at 0x010141b7 [=>                            ]   8.2% 49152/598245 bytes...
Writing at 0x0101cb18 [==>                           ]  11.0% 65536/598245 bytes...
Writing at 0x010214e7 [===>                          ]  13.7% 81920/598245 bytes...
Writing at 0x010254df [===>                          ]  16.4% 98304/598245 bytes...
Writing at 0x010294da [====>                         ]  19.2% 114688/598245 bytes...
Writing at 0x0102d4d5 [=====>                        ]  21.9% 131072/598245 bytes...
Writing at 0x010314d0 [======>                       ]  24.6% 147456/598245 bytes...
Writing at 0x010354cb [=======>                      ]  27.4% 163840/598245 bytes...
Writing at 0x010394c6 [========>                     ]  30.1% 180224/598245 bytes...
Writing at 0x0103d4c1 [========>                     ]  32.9% 196608/598245 bytes...
Writing at 0x010414bc [=========>                    ]  35.6% 212992/598245 bytes...
Writing at 0x010454b7 [==========>                   ]  38.3% 229376/598245 bytes...
Writing at 0x010494b2 [===========>                  ]  41.1% 245760/598245 bytes...
[... 189 lines cut ...]
Reading from 0x0008f000 [=========================>    ]  89.3% 585728/656136 bytes...
Reading from 0x00091000 [==========================>   ]  90.5% 593920/656136 bytes...
Reading from 0x00092000 [==========================>   ]  91.1% 598016/656136 bytes...
Reading from 0x00093000 [==========================>   ]  91.8% 602112/656136 bytes...
Reading from 0x00094000 [==========================>   ]  92.4% 606208/656136 bytes...
Reading from 0x00095000 [==========================>   ]  93.0% 610304/656136 bytes...
Reading from 0x00096000 [===========================>  ]  93.6% 614400/656136 bytes...
Reading from 0x00097000 [===========================>  ]  94.3% 618496/656136 bytes...
Reading from 0x00098000 [===========================>  ]  94.9% 622592/656136 bytes...
Reading from 0x0009a000 [===========================>  ]  96.1% 630784/656136 bytes...
Reading from 0x0009b000 [============================> ]  96.8% 634880/656136 bytes...
Reading from 0x0009c000 [============================> ]  97.4% 638976/656136 bytes...
Reading from 0x0009d000 [============================> ]  98.0% 643072/656136 bytes...
Reading from 0x0009e000 [============================> ]  98.6% 647168/656136 bytes...
Reading from 0x0009f000 [============================> ]  99.3% 651264/656136 bytes...
Reading from 0x000a0000 [============================> ]  99.9% 655360/656136 bytes...
Reading from 0x000a0308 [==============================] 100.0% 656136/656136 bytes...

Hard resetting via RTS pin...
```

And the binaries, wrote to 0x0100 0000 but read from 0x0000 0000, result identical.
Moreover if I try to read at address 0x0100 0000 the stub uses the 24bit addressing resulting in reading to 0x0000 0000.

What am I doing wrong?

RELOC-DEV, 2025-05-05:
For further information, the SPI flash connected is the Windbond W25Q01JV (1Gbit) and it is correctly detected from the ESP32-S3:

```
python.exe .\esptool.py -c esp32s3 -p COM7 -b 921600 flash-id --spi-connection 45,48,47,3,14
esptool.py v4.8.1
Loaded custom configuration from D:\2024-SICOR\ext_flash_load\esptool-git\esptool\esptool.cfg
Serial port COM7:
Connecting...
Connected to ESP32-S3 on COM7:
Chip type:          ESP32-S3 (QFN56) (revision v0.2)
Features:           Wi-Fi, BT 5 (LE), Dual Core + LP Core, 240MHz, Embedded PSRAM 8MB (AP_3v3)
Crystal frequency:  40MHz
USB mode:           USB-Serial/JTAG
MAC:                dc:da:0c:77:19:98

Uploading stub flasher...
Running stub flasher...
Stub flasher running.
Changing baud rate to 921600...
Changed.

Configuring SPI flash mode (CLK:45, Q:48, D:47, HD:3, CS:14)...
Flash Memory Information:
=========================
Manufacturer: ef
Device: 4021
Detected flash size: 128MB
Flash type set in eFuse: quad (4 data lines)
Flash voltage set by eFuse: 3.3V
```

radimkarnis, 2025-05-18:
Seeing that the flash chip gets attached and detected correctly is a sign you are not doing anything wrong. Usually, if this works, 32-bit addressing support should kick in automatically. 

The issue might be the used flash chip. The support in the stub flasher ([and across our frameworks](https://docs.espressif.com/projects/esp-idf/en/release-v5.4/esp32s3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#bit-address-support-of-qspi-flash-chips)) is tested with the following flash chips: `W25Q256`, `GD25Q256`. Although the Winbond protocol and command op-codes should theoretically be the same, there might be a functional difference with your `W25Q01JV` to enable the 4-byte addressing mode.

I will need to look into this and try to reproduce this to see what the issue is. Crrently, I don't have the `W25Q01JV` on hand, so it might take some time before I can order it or reproduce with other flash chips.

To help gather more information, if you have a logic analyzer / oscilloscope, you can try to see if the SPI flash commands are really being executed in the 4-byte mode (the address should be 4 bytes instead of just 3).

radimkarnis, 2025-05-19:
@RELOC-DEV I have been able to reproduce the issue with another flash chip. This is a bug in the stub flasher - if the flash chip is connected externally, it is not getting recognized as >16MB and the 32-bit addressing support doesn't turn on (see [this line](https://github.com/espressif/esptool-legacy-flasher-stub/blob/master/flasher_stub/stub_flasher.c#L550)). I don't have a fix yet, but if I force the support on manually, it works.

I have updated the issue name accordingly.

RELOC-DEV, 2025-05-20:
> [@RELOC-DEV](https://github.com/RELOC-DEV) I have been able to reproduce the issue with another flash chip. This is a bug in the stub flasher - if the flash chip is connected externally, it is not [...]

Great news!
Thank you for the investigation!

Could you please share the generated .json stubs file with the forced 32-bit addressing so that I can try on my side either in the meanwhile?
Thank you!

radimkarnis, 2025-05-20:
Sure, feel free to test this "hacked" stub file: [esp32s3.json](https://github.com/user-attachments/files/20344418/esp32s3.json)

RELOC-DEV, 2025-05-20:
I confirm that I am now able to load all the 128MB memory!

I'll wait for the official fix but now I have a workaround.

Thank you again for your time.

radimkarnis, 2025-05-20:
Glad I could help at least with this quick "patch" and thanks for the confirmation!

I will keep this open until we issue a proper fix.