espressif/esptool #1087: New release

martijnbuts, 2025-05-27:
### Is your feature request related to a problem?

I want to use a resent release to test with flashing esp32 p4

### Describe the solution you'd like

New release,  recent release is almost a year old

radimkarnis, 2025-05-27:
Hello @martijnbuts,

The ESP32-P4 target has been fully supported in esptool since version 4.8.0, which is available.

If you want to try the latest improvements, you can install development versions from PyPI:
- latest v4: `pip install esptool==4.9.dev8`
- latest v5: `pip install esptool==5.0.dev0`

If you want to use the pre-built executables, you can try any of the latest artifacts from the GitHub Actions pipeline, e.g. https://github.com/espressif/esptool/actions/runs/15163935346.

Sorry for the wait, we are working hard to release v5 very soon!

radimkarnis, 2025-05-29:
I will close this ticket, thanks for reaching out!

closed 2025-05-29 by radimkarnis