espressif/esptool #1091: esptool.py send command (ESPTOOL-1077)

brekemeier, 2025-06-10:
### Operating System

windows 11 / esp-idf

### Esptool Version

4.0

### Python Version

3.13

### Full Esptool Command Line that Was Run

esptool.py -p COM-Port  flash-id

### Esptool Output

```shell
Error: make flash-id // all 
esptool -p COMPORT flash_id 
please remove in online-Document
```

Dzarda7, 2025-06-10:
Hello @brekemeier, I am not really sure what you are trying to accomplish, but if you are using esptool v4.0, it does not use commands with dash like flash-id, only underscore ones like flash_id.

brekemeier, 2025-06-10:
Hallo, 

 

is not a Problem, the Help-Document: esptool.py –-port  COMx  !!! erase-flash !!! #All commend is not correct erase_flash is true! Please correct in the Help Doc (- with _) .

thanks

brekemeier, 2025-06-10:
In the Document, try: comment-id, is not correct . Please must  comment_id for all comments = help Document!

radimkarnis, 2025-06-10:
@brekemeier command format was changed from `flash_id` -> `flash-id` in esptool `v5` and that is also reflected in the latest documentation.

Esptool `v4` (which you are using) uses commands with underscores only (e.g., `flash_id`). If you switch the [documentation](https://docs.espressif.com/projects/esptool/en/release-v4/esp32/index.html) to `release/v4`, you will see the correct command format for your case.

[image]

brekemeier, 2025-06-11:
Thangs

closed 2025-06-11 by radimkarnis