espressif/esptool #1106: Invalid head of packet during uploading code

cniedzi, 2025-07-28:
### Operating System

Windows 11

### Esptool Version

5.0.1

### Python Version

Python 3.13.5

### Chip Description

ESP32-D0WD-V3 (revision v3.1)

### Device Description

Plain ESP32 Dev Module with CP2102 USB to UART chip.

Upload works seamlessly when used with ESP32 Dev Module based on CH340, or with Esptool 5.0.0.-dev1.

### Hardware Configuration

Plain module without any connection. Only USB cable connected to PC for uploading code.

### How is Esptool Run

Windows CMD

### Full Esptool Command Line that Was Run

C:\Users\User>C:\Users\User\.platformio\penv\Scripts\esptool.exe --chip esp32 --port "COM3" --baud 921600 --before default-reset --after hard-reset write-flash -z --flash-mode dio --flash-freq 80m --flash-size detect 0x0 "D:\test\bootloader.bin" 0x8000 "D:\test\partitions.bin" 0xe000 "D:\test\boot_app0.bin" 0x10000 "D:\test\firmware.bin"

### Esptool Output

```shell
C:\Users\User>C:\Users\User\.platformio\penv\Scripts\esptool.exe version
esptool v5.0.1
5.0.1

C:\Users\User>C:\Users\User\.platformio\penv\Scripts\esptool.exe --chip esp32 --port "COM3" --baud 921600 --before default-reset --after hard-reset write-flash -z --flash-mode dio --flash-freq 80m --flash-size detect 0x0 "D:\test\bootloader.bin" 0x8000 "D:\test\partitions.bin" 0xe000 "D:\test\boot_app0.bin" 0x10000 "D:\test\firmware.bin"
esptool v5.0.1
Serial port COM3:
Connecting.......

A fatal error occurred: Invalid head of packet (0x00): Possible serial noise or corruption.
```

### More Information

Upload works seamlessly when used with ESP32 Dev Module based on **CH340**, or with **Esptool 5.0.0-dev1**.

**Esptool 5.0.1**

```
C:\Users\User>C:\Users\User\.platformio\penv\Scripts\esptool.exe version
esptool v5.0.1
5.0.1

C:\Users\User>C:\Users\User\.platformio\penv\Scripts\esptool.exe --chip esp32 --port "COM3" --baud 921600 --before default-reset --after hard-reset write-flash -z --flash-mode dio --flash-freq 80m --flash-size detect 0x0 "D:\test\bootloader.bin" 0x8000 "D:\test\partitions.bin" 0xe000 "D:\test\boot_app0.bin" 0x10000 "D:\test\firmware.bin"
esptool v5.0.1
Serial port COM3:
Connecting.......

A fatal error occurred: Invalid head of packet (0x00): Possible serial noise or corruption.
```

**Esptool 5.0.0.-dev1**

```
C:\Users\User>C:\Users\User\.platformio\penv\Scripts\esptool.exe version
esptool.py v5.0.0-dev1
5.0.0-dev1

C:\Users\User>C:\Users\User\.platformio\penv\Scripts\esptool.exe --chip esp32 --port "COM3" --baud 921600 --before default-reset --after hard-reset write-flash -z --flash-mode dio --flash-freq 80m --flash-size detect 0x0 "D:\test\bootloader.bin" 0x8000 "D:\test\partitions.bin" 0xe000 "D:\test\boot_app0.bin" 0x10000 "D:\test\firmware.bin"
esptool.py v5.0.0-dev1
Connected to ESP32 on COM3:
Chip type:          ESP32-D0WD-V3 (revision v3.1)
Features:           Wi-Fi, BT, Dual Core + LP Core, 240MHz, Vref calibration in eFuse, Coding Scheme None
Crystal frequency:  40MHz
MAC:                d0:ef:76:46:60:6c

Stub flasher running.
Changing baud rate to 921600...
Changed.

Writing at 0x0006d3ff [█████░░░░░░░░░░░░░░░░░░░░░░░░░]  20.0% 163840/819568 bytes...
Writing at 0x00067780 [█████░░░░░░░░░░░░░░░░░░░░░░░░░]  18.0% 147456/819568 bytes...
Writing at 0x0005fc9b [████░░░░░░░░░░░░░░░░░░░░░░░░░░]  16.0% 131072/819568 bytes...
Writing at 0x00059978 [████░░░░░░░░░░░░░░░░░░░░░░░░░░]  14.0% 114688/819568 bytes...
Writing at 0x00053e5b [███░░░░░░░░░░░░░░░░░░░░░░░░░░░]  12.0% 98304/819568 bytes...
Writing at 0x00047204 [██░░░░░░░░░░░░░░░░░░░░░░░░░░░░]  10.0% 81920/819568 bytes...
Writing at 0x0003bbe9 [██░░░░░░░░░░░░░░░░░░░░░░░░░░░░]   8.0% 65536/819568 bytes...
Writing at 0x00033e73 [█░░░░░░░░░░░░░░░░░░░░░░░░░░░░░]   6.0% 49152/819568 bytes...
Writing at 0x0002ac1e [█░░░░░░░░░░░░░░░░░░░░░░░░░░░░░]   4.0% 32768/819568 bytes...
Writing at 0x0001d606 [░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░]   2.0% 16384/819568 bytes...
Writing at 0x00010000 [░░░░░░░░░░░░░░░░░░░░░░░░░░░░░░]   0.0% 0/819568 bytes...

```

### I Have Read the Troubleshooting Guide

- [x] I confirm I have read the troubleshooting guide.

Dzarda7, 2025-07-28:
Hi @cniedzi, I see that you are using some modified esptool from platformio. I am unable to reproduce your issue with esptool installed using pip. Can you please try to either install using pip as described [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html) or just download binary from here and try it with official esptool? Thanks.

If the issue permits, please try to update drivers for the cp2102 using automatic update in Windows as described [here](https://support.microsoft.com/en-us/windows/update-drivers-through-device-manager-in-windows-ec62f46c-ff14-c91d-eead-d7126dc1f7b6), I encountered the similar issue before and this solved it. Thanks.

cniedzi, 2025-07-28:
I'm not super experienced with Git, so I'm not sure what to do next... :(

```
PS C:\Users\Czarek\Documents\esptool\esptool> pip install --user -e .
Obtaining file:///C:/Users/Czarek/Documents/esptool/esptool
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: bitstring!=4.2.0,>=3.1.6 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (4.3.1)
Requirement already satisfied: cryptography>=43.0.0 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (45.0.5)
Requirement already satisfied: pyserial>=3.3 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (3.5)
Requirement already satisfied: reedsolo<1.8,>=1.5.3 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (1.7.0)
Requirement already satisfied: PyYAML>=5.1 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (6.0.2)
Requirement already satisfied: intelhex in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (2.3.0)
Requirement already satisfied: rich_click in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (1.8.9)
Requirement already satisfied: click<8.2.0 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from esptool==5.0.1) (8.1.8)
Requirement already satisfied: colorama in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from click<8.2.0->esptool==5.0.1) (0.4.6)
Requirement already satisfied: bitarray<4.0,>=3.0.0 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from bitstring!=4.2.0,>=3.1.6->esptool==5.0.1) (3.5.2)
Requirement already satisfied: cffi>=1.14 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from cryptography>=43.0.0->esptool==5.0.1) (1.17.1)
Requirement already satisfied: pycparser in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from cffi>=1.14->cryptography>=43.0.0->esptool==5.0.1) (2.22)
Requirement already satisfied: rich>=10.7 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from rich_click->esptool==5.0.1) (14.1.0)
Requirement already satisfied: typing_extensions>=4 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from rich_click->esptool==5.0.1) (4.14.1)
Requirement already satisfied: markdown-it-py>=2.2.0 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from rich>=10.7->rich_click->esptool==5.0.1) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from rich>=10.7->rich_click->esptool==5.0.1) (2.19.2)
Requirement already satisfied: mdurl~=0.1 in c:\users\czarek\appdata\local\programs\python\python313\lib\site-packages (from markdown-it-py>=2.2.0->rich>=10.7->rich_click->esptool==5.0.1) (0.1.2)
Building wheels for collected packages: esptool
  Building editable for esptool (pyproject.toml) ... done
  Created wheel for esptool: filename=esptool-5.0.1-0.editable-py3-none-any.whl size=11437 sha256=c65adb96db53956b219698286a727c85725bc008fb1a4b1fd45448e880442fe3
  Stored in directory: C:\Users\Czarek\AppData\Local\Temp\pip-ephem-wheel-cache-04ddseqx\wheels\8a\85\5e\19cb83b6f20982baea4e173a51c25bf8aefdd3e30298d37758
Successfully built esptool
Installing collected packages: esptool
  Attempting uninstall: esptool
    Found existing installation: esptool 5.0.1
    Uninstalling esptool-5.0.1:
      Successfully uninstalled esptool-5.0.1
  WARNING: The scripts esp_rfc2217_server.exe, espefuse.exe, espsecure.exe and esptool.exe are installed in 'C:\Users\Czarek\AppData\Roaming\Python\Python313\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed esptool-5.0.1
```

Dzarda7, 2025-07-28:
This does not build the exe, this installs the esptool as a python package so you can run it now as `esptool.py <some_command>`.  Now you need to be in the esptool folder you installed using `pip install --user -e .`.  [Here](https://www.geeksforgeeks.org/git/git-bisect/) is a nice example of how to use `git bisect`. You basically run `git bisect start`, then specify good and bad commit. (for latest commit `git bisect bad 0f32306551ae35fbaf2a6ee4a28e56d59f41e1ae`, for release 4.9.0dev5 commit `git bisect good a32988e2d5f02845ce16e22022f5b64368f12572`). Now git will automatically go through the history of changes and let you find the place where it stopped working. After bad and good is specified, you run `esptool.py flash_id`, if it works, you write `git bisect good`, if not `git bisect bad`. After few steps, git will print the commit which caused the issue.

cniedzi, 2025-07-28:
> This does not build the exe, this installs the esptool as a python package so you can run it now as `esptool.py <some_command>`. Now you need to be in the esptool folder you installed using `pip [...]

Thx, will try!

Dzarda7, 2025-07-28:
The version report might be incorrect in the commits in between releases. This one is not present in v5.0.0dev1, the first release where this commit is present is v5.0.0. Thanks for the investigation. I will try to reproduce it and fix it. The issue is present only on cp2102 right? I encountered this exact issue with this commit, but it was a driver issue. Can you please try to run following commands on the not working version (or commit) and paste the output here:

`esptool.py --trace flash_id`
`esptool.py --trace --chip esp32 flash_id`

Dzarda7, 2025-07-29:
Hello @cniedzi, I found the issue, the reason is that some version of CP2102 driver for Windows most probably does not properly flush the data properly. New version should solve it I believe, but to avoid future issues with this and other drivers I prepared the fix that will drain the data. Unfortunately I am not able to reproduce the issue still even on multiple Windows machines. Can you please test it? I added the fix to my for [here](https://github.com/Dzarda7/esptool). Again `git clone` and `pip install --user -e .` in the newly cloned folder (it should reinstall the previously installed esptool). Thank you very much.

cniedzi, 2025-07-29:
Hi @Dzarda7, looks good.

```
D:\xxx>esptool.py --chip esp32 --port "COM3" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0 "D:\test\bootloader.bin" 0x8000 "D:\test\partitions.bin" 0xe000 "D:\test\boot_app0.bin" 0x10000 "D:\test\firmware.bin"
Warning: DEPRECATED: 'esptool.py' is deprecated. Please use 'esptool' instead. The '.py' suffix will be removed in a future major release.
Warning: Deprecated: Option '--flash_mode' is deprecated. Use '--flash-mode' instead.
Warning: Deprecated: Option '--flash_freq' is deprecated. Use '--flash-freq' instead.
Warning: Deprecated: Option '--flash_size' is deprecated. Use '--flash-size' instead.
Warning: Deprecated: Choice 'default_reset' for option '--before' is deprecated. Use 'default-reset' instead.
Warning: Deprecated: Choice 'hard_reset' for option '--after' is deprecated. Use 'hard-reset' instead.
Warning: Deprecated: Command 'write_flash' is deprecated. Use 'write-flash' instead.
esptool v5.0.1
Connected to ESP32 on COM3:
Chip type:          ESP32-D0WD-V3 (revision v3.1)
Features:           Wi-Fi, BT, Dual Core + LP Core, 240MHz, Vref calibration in eFuse, Coding Scheme None
Crystal frequency:  40MHz
MAC:                d0:ef:76:46:60:6c

Stub flasher running.
Changing baud rate to 921600...
Changed.

Configuring flash size...
Auto-detected flash size: 4MB
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00164fff...
Wrote 23696 bytes (15164 compressed) at 0x00000000 in 0.3 seconds (568.5 kbit/s).
Hash of data verified.
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (1371.7 kbit/s).
Hash of data verified.
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (1942.4 kbit/s).
Hash of data verified.
Wrote 1396400 bytes (819568 compressed) at 0x00010000 in 11.5 seconds (968.6 kbit/s).
Hash of data verified.

Hard resetting via RTS pin...
```

cniedzi, 2025-07-29:
> Did you update the drivers in between or changed something? Do you know why the built one works suddenly?

No, I haven't touched anything since yesterday.

Dzarda7, 2025-07-29:
Maybe it needed reboot after driver update from yesterday.

dobairoland, 2025-07-29:
@Dzarda7 I believe this `esptool.exe` is used: `\python_env\idf6.0_py3.11_env\Scripts\esptool.exe`. This is not the self-contained pyinstaller from the release.

cniedzi, 2025-07-29:
> Yes, this one was installed by pip:
> 
> ```
> C:\Users\Czarek\AppData\Roaming\Python\Python313\Scripts/esptool.exe
> ```

yes, and this one works for me, whie the official one not.

Dzarda7, 2025-07-29:
Okay, thanks. I did not know that pip somehow creates exe on Windows. Thanks, the fix works then. Will try to merge it soon.