espressif/esptool #1107: ModuleNotFoundError: No module named 'esptool.__init__'; 'esptool' is not a package

franz-ms-muc, 2025-08-01:
### Operating System

Ubuntu

### Esptool Version

--

### Python Version

--

### Full Esptool Command Line that Was Run

esptool version

### Esptool Output

```shell
franz@franz-exone-Business-Tower-X15:~/git/hr/ConnectID2/ConnectID2/build_ESP32$ esptool.py version
Traceback (most recent call last):
  File "/home/franz/.espressif/python_env/idf5.5_py3.13_env/bin/esptool.py", line 3, in <module>
    from esptool.__init__ import _main
  File "/home/franz/.espressif/python_env/idf5.5_py3.13_env/bin/esptool.py", line 3, in <module>
    from esptool.__init__ import _main
ModuleNotFoundError: No module named 'esptool.__init__'; 'esptool' is not a package
```

dobairoland, 2025-08-01:
@franz-ms-muc Thank you for your report. I've got similar ones from others and I'm working on this right now but could not yet figure it out. Could you please help me collect more information about your system and why it is happening?

I assume that upgrade helped you because installed 5.0.2. Can you please try the following versions and see if they have the above issue?

```sh
pip install esptool==4.9.1
pip install esptool==4.10.dev1
```

When you get the above error, could you please run the following commands?

```sh
which esptool.py
esptool version
python -m esptool version
```

dhalbert, 2025-08-01:
I've narrowed this down to a problem with 4.9.1, which causes the error above.
4.9.0 works.
4.10.dev1 works.
This is in a build environment with ESP-IDF 5.4.1.

This is affecting our CI jobs, so I'm hoping you're going to do a 4.10 release soon (or backport of something to 4.9.2). Thanks!

dhalbert, 2025-08-01:
The specific error with 4.9.1 is on this line in our Makefile:
```
	$(Q)esptool.py --chip $(IDF_TARGET) elf2image $(FLASH_FLAGS) --elf-sha256-offset 0xb0 -o $@ $^
```
so something with `elf2image`, apparently.

dobairoland, 2025-08-05:
https://dl.espressif.com/pypi/esptool/esptool-4.9.1-py3-none-any.whl was removed as well.

mattj103, 2026-01-29:
@dobairoland This has happened again. Our builds started failing today. From what I can tell the difference is

2026-01-27T20:12:11.7437916Z   Downloading esptool-4.11.0.tar.gz (471 kB)

VS Today 

2026-01-29T17:24:32.7794452Z   Downloading https://dl.espressif.com/pypi/esptool/esptool-4.11.0-py3-none-any.whl (597 kB)

dhalbert, 2026-01-29:
Echoing @mattj103: same thing, e.g.: https://github.com/adafruit/circuitpython/actions/runs/21492114191/job/61927441330?pr=10786