espressif/esptool #1122: esptool crash on boot

tbwcjw, 2025-09-21:
### Operating System

Ubuntu 20.04

### Esptool Version

v4.7.0

### Python Version

Python 3.12.3

### Full Esptool Command Line that Was Run

/usr/bin/python3 /usr/bin/esptool chip_id

### Esptool Output

```shell
Traceback:
 Traceback (most recent call last):
   File "/usr/bin/esptool", line 37, in <module>
     esptool._main()
   File "/usr/lib/python3/dist-packages/esptool/__init__.py", line 1139, in _main
     main()
   File "/usr/lib/python3/dist-packages/esptool/__init__.py", line 751, in main
     esp = esp.run_stub()
           ^^^^^^^^^^^^^^
   File "/usr/lib/python3/dist-packages/esptool/loader.py", line 996, in run_stub
     stub = StubFlasher(get_stub_json_path(self.CHIP_NAME))
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/lib/python3/dist-packages/esptool/loader.py", line 159, in __init__
     with open(json_path) as json_file:
          ^^^^^^^^^^^^^^^
 FileNotFoundError: [Errno 2] No such file or directory: '/usr/lib/python3/dist-packages/esptool/targets/stub_flasher/stub_flasher_32.json'
```

### More Information

On boot, I got a "Report a problem..." window. The above traceback is from /var/crash/_usr_bin_esptool.0.crash.

tbwcjw, 2025-09-22:
The bug isn't so much the output of esptool, it's *why* esptool ran on boot in the first place. it's not a service, it has no cronjobs, and as far as I know there should be no daemon running at boot time.

tbwcjw, 2025-09-22:
Ah! Yes you're right. I did install with apt, discovered it didn't work, then moved on to the pip one, forgetting to uninstall it. Must be a Ubuntu side type thing. I'll get more info and send a bug report over that way. Thanks. It didn't occur to me this could be the case as the command line arg was /usr/bin/python3.

dobairoland, 2025-09-22:
Thanks for the confirmation. I'm closing this because we are not able to fix the package installed with `apt`.

closed 2025-09-22 by dobairoland