espressif/esptool #1125: `sign-data` command requires `key` argument in `--hsm` mode

ulrichb, 2025-09-26:
### Operating System

Ubuntu

### Esptool Version

5.1.0

### Python Version

Python 3

### Full Esptool Command Line that Was Run

espsecure sign-data --version 2 --hsm --hsm-config "hsm_config.ini" ...

### Esptool Output

```shell
Missing option '--keyfile' / '-k'.
```

### What is the Expected Behaviour?

Don't require a `--keyfile` parameter (currently I'm providing a dummy file) when using the `--hsm` command line ([as documented](https://docs.espressif.com/projects/esptool/en/latest/esp32/espsecure/index.html#remote-signing-using-an-external-hsm)).

dobairoland, 2025-09-26:
@ulrichb Could you please try the latest master version? I think this might have already been fixed by https://github.com/espressif/esptool/commit/ee35dda6994af180f81ffa24ec5da40b1cf4ba40 which is not yet released.

ulrichb, 2025-09-26:
Oh just 3 days ago :)

Yes, according to the code this is exactly what this ticket is about.

Sorry for the dumb question, but I don't know how to test a master version (I just did `pip install 'esptool[hsm]'`).

But nevermind, I'm very sure that the commit you referenced will fix my issue. So many thanks!