espressif/esptool #1126: ESP Tool fails consistently on one computer and works on another with ESP32-Pi board from Sequent Microsystems

tjhayko, 2025-09-26:
### Operating System

Windows 11

### Esptool Version

5.1.0

### Python Version

Python 3.11.0

### Chip Description

ESP32 (https://sequentmicrosystems.com/products/esp32-pi-low-cost-replacement-for-raspberry-pi?_pos=1&_psq=ESP32&_ss=e&_v=1.0)

### Device Description

None.

### Hardware Configuration

https://sequentmicrosystems.com/products/eight-mosfets-8-layer-stackable-card-for-raspberry-pi (i2c device)

### How is Esptool Run

Windows Terminal (although same result from Arduino IDE or Visual Studio Code with Platform.IO)

### Full Esptool Command Line that Was Run

"D:\old-pc\d-drive\Software\esptool\esptool-windows-amd64\esptool" -t --chip esp32 --port "COM5" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 "D:\Dev\8MOSFET Dew Heater\ESP32-Dew-Heater\.pio\build\esp32doit-devkit-v1\bootloader.bin" 0x8000 "D:\Dev\8MOSFET Dew Heater\ESP32-Dew-Heater\.pio\build\esp32doit-devkit-v1\partitions.bin" 0xe000 C:\Users\tomha\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 0x10000 .pio\build\esp32doit-devkit-v1\firmware.bin

### Esptool Output

```shell
output logs are large.  One from a machine where it works properly most of the time, the other from the one that always fails.
```

### More Information

[2025-09-26-working.txt](https://github.com/user-attachments/files/22568835/2025-09-26-working.txt)

[2025-09-26 - Not working.txt](https://github.com/user-attachments/files/22568836/2025-09-26.-.Not.working.txt)

### I Have Read the Troubleshooting Guide

- [x] I confirm I have read the troubleshooting guide.

jedi7, 2025-09-30:
@tjhayko I primarily use a USB oscilloscope for debugging because it allows me to see the voltage levels. However, the best tool depends on your specific use case, so I can't give a definitive recommendation.

Looking at your traces, I can see that even on the old computer, it takes a longer time to switch into download mode. When I run the programming sequence myself, I typically see just one sync sequence (the "UUUUU" strings) before programming starts successfully.

In your traces, however, I see multiple failed sync attempts. The tool is sending the sync command, but the ESP32 isn't in download mode to respond correctly. Instead, we're seeing the serial output from the already-running firmware.

As a reminder, entering download mode is triggered by toggling the RTS and DTR lines (see the [Espressif boot mode documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html)). This is the point where the process is failing.

From this evidence, it seems like a hardware-related issue. A common culprit is excessive capacitance on the RTS/DTR lines, which can slow down or distort the control signals, preventing a clean entry into the bootloader. It's hard to say for sure without probing the board.

Have you found any relevant discussions in the product's forums or issue trackers that might point to a known hardware design flaw?

tjhayko, 2025-09-30:
Would a 50Mhz bandwidth oscilloscope be workable for this purpose?  I'm trying to get something useful enough, but not overspend.  

If I can capture the same information with a logic analyzer, I'll go with one of those as they are significantly cheaper.

dobairoland, 2025-10-06:
Please ping us if this proves to be an esptool issue.