espressif/esptool #1127: 必须进入下载模式才能烧录程序, 以前不是这样的

Tinyu-Zhao, 2025-09-29:
### Operating System

macOS 15.4.1

### Esptool Version

esptool.py v4.9.0

### Python Version

Python 3.13.3

### Chip Description

ESP32S3

### Device Description

[M5StackCoreS3](https://docs.m5stack.com/en/core/CoreS3)

### Hardware Configuration

https://m5stack-doc.oss-cn-shenzhen.aliyuncs.com/490/Sch_M5_CoreS3_v1.0.pdf

### How is Esptool Run

No IDE

### Full Esptool Command Line that Was Run

esptool.py --chip esp32s3 --port /dev/cu.usbmodem211101 --baud 1500000 write_flash 0x0 /Users/tinyu/Developer/Code/uiflow_micropython/uiflow_develop/m5stack/build-M5STACK_CoreS3/uiflow-885edad6.bin

### Esptool Output

```shell
~ esptool.py --chip esp32s3 --port /dev/cu.usbmodem211101 --baud 1500000 write_flash 0x0 /Users/tinyu/Developer/Code/uiflow_micropython/uiflow_develop/m5stack/build-M5STACK_CoreS3/uiflow-885edad6.bin
esptool.py v4.9.0
Serial port /dev/cu.usbmodem211101
Connecting....

A serial exception error occurred: Could not configure port: (6, 'Device not configured')
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

### More Information

➜  ~ ls /dev/cu.*
/dev/cu.Bluetooth-Incoming-Port /dev/cu.usbmodem211101
/dev/cu.debug-console
---------------------------
Before Run esptool write_flash command
----------------------------
➜  ~ esptool.py --chip esp32s3 --port /dev/cu.usbmodem211101 --baud 1500000 write_flash 0x0 /Users/tinyu/Developer/Code/uiflow_micropython/uiflow_develop/m5stack/build-M5STACK_CoreS3/uiflow-885edad6.bin
esptool.py v4.9.0
Serial port /dev/cu.usbmodem211101
Connecting....

A serial exception error occurred: Could not configure port: (6, 'Device not configured')
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
---------------------------
After Run esptool write_flash command
----------------------------
➜  ~ ls /dev/cu.*
/dev/cu.Bluetooth-Incoming-Port /dev/cu.debug-console

### I Have Read the Troubleshooting Guide

- [x] I confirm I have read the troubleshooting guide.

radimkarnis, 2025-09-29:
Hello @Tinyu-Zhao,

can you please specify what you mean by `this wasn't the case before`?
Have you recently updated esptool, or has this started with a new build of your app?

If the app reconfigures the GPIO pins used for the USB communication, the port can become unusable (until download mode is entered manually, then the ROM reconfigures the pins back to a usable USB-CDC device).

The fact that the port disappears after you run esptool signals either to faulty HW configuration (mentioned above) or faulty USB drivers. After opening the port fails, it seems to get de-enumerated because the OS deems it unusable.

This is definitely not an issue in esptool. It is dependent on a working serial connection. In this case, the serial port cannot even be opened.

Tinyu-Zhao, 2025-09-29:
> Hello [@Tinyu-Zhao](https://github.com/Tinyu-Zhao),
>
> can you please specify what you mean by `this wasn't the case before`? Have you recently updated esptool, or has this started with a new build of [...]

Previously, I could use esptool to flash the ESP32S3 without putting it into download mode.

I haven't updated esptool recently, and I'm getting errors whether I use the esptool built into esp-idf or the one installed via `pip install esptool`.

However, if I use `idc.py flash`, this issue won't occur.

this is hardware sch
[image]

dobairoland, 2025-09-29:
> However, if I use idc.py flash, this issue won't occur.

Also please let us know what esptool version is printed there.

Tinyu-Zhao, 2025-09-29:
> I just reproduced it again
> 
> 1. Re-power on the device that just failed to flash via esptool.py and flash the esp-idf/examples/get-started/hello_world program using `idf.py flash` ----> I get the same error
> 
> <img alt="Image" width="1433" height="561" src="https://private-user-images.githubusercontent.com/44396641/495082532-4857ff11-a003-4b4e-9993-a4d9c452e00b.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxMzY1NTIsIm5iZiI6MTc1OTEzNjI1MiwicGF0aCI6Ii80NDM5NjY0MS80OTUwODI1MzItNDg1N2ZmMTEtYTAwMy00YjRlLTk5OTMtYTRkOWM0NTJlMDBiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDA4NTczMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2YjNiMWIwNzI1YzcxMDVmYTYwOGQ0NjZiNDlkYTM2YTVmNTdmMDhlMGZjNjc2NjRhYjU4MDkwMDY0ZTRkNDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.jUhKNkKF1cpz-saknpggUpHNjrzAI7I3QfDnqjPMEbA">
> 2. After the device enters download mode, use `idf.py flash`` to burn the program again. --> Burn successfully
> 
> <img alt="Image" width="1420" height="636" src="https://private-user-images.githubusercontent.com/44396641/495082879-181758cf-3e6b-454c-88ca-a79099c22949.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxMzY1NTIsIm5iZiI6MTc1OTEzNjI1MiwicGF0aCI6Ii80NDM5NjY0MS80OTUwODI4NzktMTgxNzU4Y2YtM2U2Yi00NTRjLTg4Y2EtYTc5MDk5YzIyOTQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDA4NTczMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3OTJhYzFlMzVhNGUzM2EyMmJlYmE3Y2RlZTM5NGFjNmZhMjAzZjhiMmEyMjA0MTM3ZTU4YmUxNzViOGFhZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CGTiyqY4Zv2TlMMti_lC29ZdvsrM3Ma0jSrfUQdb5-U">
> 3. Without putting the device into download mode, execute `idf.py flash` to flash the program. The flash is successful (I have tried multiple times and all are successful).
> 4. Without putting the device into download mode, execute `esptool.py -p /dev/cu.usbmodem211101 -b 1500000 write_flash 0x0 ~/Developer/Code/uiflow_micropython/uiflow_develop/m5stack/build-M5STACK_CoreS3/uiflow-4eb4f152.bin` --> Burn successfully
> 
> <img alt="Image" width="1431" height="421" src="https://private-user-images.githubusercontent.com/44396641/495083589-da3c7f06-b9ce-4e3d-993c-95a6a0faad17.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxMzY1NTIsIm5iZiI6MTc1OTEzNjI1MiwicGF0aCI6Ii80NDM5NjY0MS80OTUwODM1ODktZGEzYzdmMDYtYjljZS00ZTNkLTk5M2MtOTVhNmEwZmFhZDE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDA4NTczMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1Nzk4MDA3ODU0ZTllY2JkZDJlZjNkZjkwNDUyOTk1YjQ3Nzg1NTY1ZTVmNjA4Y2NkYzhiY2YxNWY2ZGU3MTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.z-qseCY64EVU6p_vnEUo9n1c6JeP8CIXH-m5s9bkmRA">
> 5. After flashing the firmware, without putting the device into download mode, execute `idf.py flash` flash program --> Flashing fails, and the same error message as Figure 1 is displayed
> 
> I don't know why this happened

> > However, if I use idc.py flash, this issue won't occur.
> 
> This command still runs the esptool that's bundled with ESP-IDF, but you say ` I'm getting errors whether I use the esptool built into esp-idf`. Can you check if there is a difference between esptool version or in the way esptool is invoked (you can see the full esptool command in the idf.py output log)?
> 
> You can also try adding the `--before usb_reset` argument (after `esptool.py`).

If `--before usb_reset` is used, no data will be received. However, there is an improvement. After executing this command, my serial port will not disappear.

radimkarnis, 2025-09-29:
Please see that the switching into download mode:
1) Works when the "hello world" app is loaded
2) Stops working when you load your app

This means your app is compiled without the USB-Serial/JTAG support, or it reconfigures the USB pins (GPIO19 and 20). Can you please check that? Without this support activated the USB-Serial/JTAG HW cannot detect and act on the reset sequence.

Tinyu-Zhao, 2025-09-29:
> Please see that the switching into download mode:
>
> 1. Works when the "hello world" app is loaded
> 2. Stops working when you load your app
>
> This means your app is compiled without the USB-Serial/JTAG [...]

🤔My SDKConfig has USB JTAG enabled, and I haven't changed the USB pins.

[image]

But I think this might indeed be a viable approach. The affected firmware is an extension we built on top of MicroPython, and the ESP32 portion of MicroPython is also an IDF project.

closed 2025-09-30 by Tinyu-Zhao